ROBOTIS-GIT/turtlebot3 #443: Turtlebot Add-on - Raspberry pi Fish-eye camera

nonso1234, 2019-06-25:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

      [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

      [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

      [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    -Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - HERE
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

 The particular pi camera recommended on the turtlebot3 manual as an add-on is currently out of stock on amazon.com  . However I found an alternative (https://www.amazon.com/Miuzei-Raspberry-Megapixels-Wide-Angel-Compatible/dp/B07BTP8VXL/ref=cm_cr_arp_d_product_sims?ie=UTF8&th=1) and I wanted to know if it works with the Turtlebot3 SBC. Thanks a lot

ROBOTIS-Kayman, 2019-06-26:
I'm sorry but we can not guarantee the operation because we do not have it.

Regards, 
Kayman

nonso1234, 2019-06-26:
ok thanks

closed 2019-06-26 by nonso1234