ROBOTIS-GIT/turtlebot3 #445: wheel_left_link and wheel_right_link not working in rviz while using docker

ubtjoe, 2019-07-01:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [x] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [x] etc (PLEASE, WRITE DOWN YOUR OS)
*I am using a docker image based on Xenial*

5. Write down software version and firmware version

    - Software version: [1.2.0]
    - Firmware version: [1.2.3]
 
6. Write down the commands you used in order

On remote pc:
```bash
roscore -p 11311
```
On turtlebotpi:
```bash
roslaunch turtlebot3_bringup turtlebot3_robot.launch
```

On remote pc:
```bash
roslaunch turtlebot3_bringup turtlebot3_remote.launch
rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
```

* I have repeatedly checked that the `$ROS_HOSTNAME` and `$ROS_MASTER_URI` are set correctly on each machine according to your [docs](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup) so I don't know what is wrong*

7. Copy and Paste your error message on terminal

    There was no error in the terminal, rviz just failed to show the lidar points and there were errors shown for the wheels.  The screenshot looked like this:

[image]
  
8. Please, describe detailedly what difficulty you are in 

I have made sure that my docker container is using the host's network, that the times are synced between the turtlebot and the docker container, that the ros install follows the same steps outlined in your manual, and that I have followed through all of your setup instructions and I am still having this problem.  Running `ifconfig` inside both the docker container and the pi show that the IP addresses are correct.  Can someone please help me?  I am running out of things to try...

RyanJaehyunShim, 2019-07-02:
@ubtjoe 

Would you test below and let us know if it works?

[Remote PC]
$ roscore

[TuetlebotPC]
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

[Remote PC]
$ export TURTLEBOT3_MODEL= waffle_pi
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch

$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

Regards,
Ryan

RyanJaehyunShim, 2019-07-02:
@ubtjoe 

We think the problem occurs because of the communication problem between the docker and the robot. 
We do not support such cases but you can find a solution by searching 'docker IP setting'. 

Regards,
Ryan

ubtjoe, 2019-07-02:
Ryan,

This does not work.  I have set the TURTLEBOT3_MODEL as an environment variable for the remote pc (which is a docker container) and verified, with `echo $TURTLEBOT3_MODEL`, that it was correctly set.  Also, as I noted in my edit of the issue:  whether I use the `-p 11311` option or not when launching `roscore` from the remote pc, the instructions don't seem to work.

Regards,

Joe

Joe Dinius, Ph.D.｜Senior Research Engineer – NARD
Email: joe.dinius@ubtrobot.com
Mobile: +1 (520) 904-8244

------------------------------------------------------------------

Sent At:2019 Jul. 1 (Mon.) 17:25
To:ROBOTIS-GIT/turtlebot3 <turtlebot3@noreply.github.com>
Cc:joe.dinius <joe.dinius@ubtrobot.com>; Mention <mention@noreply.github.com>

ubtjoe, 2019-07-02:
@RyanJaehyunShim  - Are there any other diagnostic steps I can follow?  I am using the `--net=host` option, which allows the container to use the host's network, so the communication with the docker container is an unlikely issue.  Running `ifconfig` from within the running container confirms this.  I am able to ping the turtlebot from within the container and all of the other ros services start correctly.

RyanJaehyunShim, 2019-07-02:
@ubtjoe 

Would you test it not in a docker container and let me know if it works?

Regards,
Ryan

ubtjoe, 2019-07-02:
I will try tomorrow and let you know.

ubtjoe, 2019-07-02:
Actually @RyanJaehyunShim I run Ubuntu 18.04 and I don't want to go through all of the effort to build kinetic from source.  This was why I went with docker in the first place :)  I will ask a question on Stack and see if I get any hits.

RyanJaehyunShim, 2019-07-02:
@ubtjoe  Good luck :)

closed 2019-07-02 by RyanJaehyunShim

ubtjoe, 2019-07-03:
@RyanJaehyunShim got it working :)  I have created a repo with instructions how other users can do the same -> https://github.com/ubtjoe/turtlebot3-docker.  Next time you get questions, you can refer them to the repo.  Thanks for you help.

RyanJaehyunShim, 2019-07-04:
@ubtjoe 

So glad you figured out the solution and thank you so much for your contribution! 
Ryan