ROBOTIS-GIT/turtlebot3 #446: How to enable lidar for ROS2 in TB3? after wget turtlebot3_lidar.tar.bz2: 404 Not Found

yechun1, 2019-07-03:
The lidar packages had been removed recently, we are not able to follow [Turtlebot3 tutorial](http://emanual.robotis.com/docs/en/popup/turtlebot3_ros2_sbc_setting/) to setup ROS2 environment to run Lidar applications on TB3. Is there any other documents show how we could use Lidar for ros2 on TB3? Or would someone like to update the instruction of "How to setup for TurtleBot3 with ROS2" in [15.3.2 SBC setup](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2 ) section. That would be very help for our [robot_devkit ](https://github.com/intel/robot_devkit.git) integration. Thanks.

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [x] Intel NUC
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu 18.04
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    -  wget https://github.com/ROBOTIS-GIT/turtlebot3/raw/ros2/turtlebot3_lidar/turtlebot3_lidar.tar.bz2
 
7. Copy and Paste your error message on terminal

    - --2019-07-03 10:57:30--  https://github.com/ROBOTIS-GIT/turtlebot3/raw/ros2/turtlebot3_lidar/turtlebot3_lidar.tar.bz2
Proxy request sent, awaiting response... 404 Not Found
2019-07-03 10:57:32 ERROR 404: Not Found.
  
8. Please, describe detailedly what difficulty you are in 

    - HERE

routiful, 2019-07-04:
Sorry for inconvenience. 
The `turtlebot3_lidar.tar.bz2` was replaced due to update the version of MicroXRCE.
After I fix a few bug, I will update a new manual for TB3 dashing release as soon as possible.

emersonknapp, 2019-07-12:
@routiful can you just update the documentation to point at a version of this repo that still contains the version for Crystal? it seems like bad practice to break the currently documented demo - you've already pinned to specific versions of the Micro-XRCE-DDS-Agent/Client, so why not pin to a specific revision of this repository as well?

routiful, 2019-07-12:
@emersonknapp 
Sorry for late update this repo.
We are continue to cooperate to solve this issue as soon as possible.

If you want to last version(for Crystal), you can find it in here.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/24d14d772520508e409b80c43859b7020d76bb82

OpusK, 2019-07-30:
Hi, all.

I have made an OpenCR firmware update, and have made an e-manual update.
I did not completely check everything I had to check, but I did update after the simple test.
Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2).

This is a temporary update, and since most developers are currently on vacation, it is likely that developers will be able to begin detailed checks next week.

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

The updates include [hls_lfcd_lds_driver](https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver/tree/ros2) can be used in ROS 2 Dashing.

closed 2019-08-21 by routiful