ROBOTIS-GIT/turtlebot3 #448: tf error: canTransform: target_frame map does not exist.

mzl626, 2019-07-06:
I'm try turtlebot3 burger simulation on ros Kinetic，ubuntu 16.04，my command is：
roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
roslaunch turtlebot3_slam turtlebot3_cartographer.launch

My error msg is：

.. logging to /home/miaozl/.ros/log/8e8b41a0-9ff6-11e9-803b-60f677e14c77/roslaunch-miaozl-ThinkPad-T470p-10200.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://miaozl-ThinkPad-T470p:40491/

SUMMARY
========

PARAMETERS
 * /move_base/DWAPlannerROS/acc_lim_theta: 3.2
 * /move_base/DWAPlannerROS/acc_lim_x: 2.5
 * /move_base/DWAPlannerROS/acc_lim_y: 0.0
 * /move_base/DWAPlannerROS/controller_frequency: 10.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.325
 * /move_base/DWAPlannerROS/goal_distance_bias: 20.0
 * /move_base/DWAPlannerROS/latch_xy_goal_tolerance: False
 * /move_base/DWAPlannerROS/max_rot_vel: 2.75
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_trans_vel: 0.22
 * /move_base/DWAPlannerROS/max_vel_x: 0.22
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_rot_vel: 1.37
 * /move_base/DWAPlannerROS/min_trans_vel: 0.11
 * /move_base/DWAPlannerROS/min_vel_x: -0.22
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.02
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 32.0
 * /move_base/DWAPlannerROS/publish_cost_grid_pc: True
 * /move_base/DWAPlannerROS/publish_traj_pc: True
 * /move_base/DWAPlannerROS/scaling_speed: 0.25
 * /move_base/DWAPlannerROS/sim_time: 1.5
 * /move_base/DWAPlannerROS/stop_time_buffer: 0.2
 * /move_base/DWAPlannerROS/vth_samples: 40
 * /move_base/DWAPlannerROS/vx_samples: 20
 * /move_base/DWAPlannerROS/vy_samples: 0
 * /move_base/DWAPlannerROS/xy_goal_tolerance: 0.05
 * /move_base/DWAPlannerROS/yaw_goal_tolerance: 0.17
 * /move_base/base_local_planner: dwa_local_planner...
 * /move_base/conservative_reset_dist: 3.0
 * /move_base/controller_frequency: 10.0
 * /move_base/controller_patience: 15.0
 * /move_base/global_costmap/cost_scaling_factor: 3.0
 * /move_base/global_costmap/footprint: [[-0.105, -0.105]...
 * /move_base/global_costmap/global_frame: /map
 * /move_base/global_costmap/inflation_radius: 1.0
 * /move_base/global_costmap/map_type: costmap
 * /move_base/global_costmap/observation_sources: scan
 * /move_base/global_costmap/obstacle_range: 3.0
 * /move_base/global_costmap/publish_frequency: 10.0
 * /move_base/global_costmap/raytrace_range: 3.5
 * /move_base/global_costmap/robot_base_frame: /base_footprint
 * /move_base/global_costmap/scan/clearing: True
 * /move_base/global_costmap/scan/data_type: LaserScan
 * /move_base/global_costmap/scan/marking: True
 * /move_base/global_costmap/scan/sensor_frame: base_scan
 * /move_base/global_costmap/scan/topic: scan
 * /move_base/global_costmap/static_map: True
 * /move_base/global_costmap/transform_tolerance: 0.5
 * /move_base/global_costmap/update_frequency: 10.0
 * /move_base/local_costmap/cost_scaling_factor: 3.0
 * /move_base/local_costmap/footprint: [[-0.105, -0.105]...
[... 52 lines cut ...]
[ INFO] [1562421726.846023531]: I0706 22:02:06.000000 10220 map_builder_bridge.cc:136] Added trajectory with ID '0'.
[ WARN] [1562421726.968002650, 32.700000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_footprint does not exist.. canTransform returned after 32.7 timeout was 0.1.
[ WARN] [1562421729.879558239, 35.523000000]: W0706 22:02:09.000000 10220 node.cc:857] Expected topic "imu" (trajectory 0) (resolved topic "/flat_imu") but no publisher is currently active.
[ WARN] [1562421729.879651417, 35.523000000]: W0706 22:02:09.000000 10220 node.cc:866] Currently available topics are: /move_base/current_goal,/rosout_agg,/odom,/scan,/gazebo_gui/parameter_descriptions,/gazebo_gui/parameter_updates,/imu,/gazebo/parameter_descriptions,/map,/cmd_vel,/landmark_poses_list,/submap_list,/scan_matched_points2,/joint_states,/rosout,/move_base/goal,/tf,/trajectory_node_list,/constraint_list,/gazebo/link_states,/gazebo/model_states,/clock,/gazebo/parameter_updates,
[ WARN] [1562421732.138737728, 37.708000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101 timeout was 0.1.
[ WARN] [1562421737.322747880, 42.718000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421742.484281842, 47.735000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421744.285101076, 49.494000000]: W0706 22:02:24.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421746.348802514, 51.498000000]: W0706 22:02:26.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421747.632028474, 52.747000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421748.408775605, 53.495000000]: W0706 22:02:28.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421750.457239661, 55.497000000]: W0706 22:02:30.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421752.524912687, 57.504000000]: W0706 22:02:32.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421752.785853207, 57.756000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421754.604800953, 59.517000000]: W0706 22:02:34.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421756.646263214, 61.498000000]: W0706 22:02:36.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421757.974327920, 62.774000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421758.708019949, 63.487000000]: W0706 22:02:38.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421760.813326855, 65.514000000]: W0706 22:02:40.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421762.912787191, 67.514000000]: W0706 22:02:42.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421763.195061447, 67.791000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1562421764.939470589, 69.487000000]: W0706 22:02:44.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)
[ WARN] [1562421767.037214080, 71.516000000]: W0706 22:02:47.000000 10220 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)

please help me!

ROBOTIS-Kayman, 2019-07-08:
Hi,
Could you refer to this issue(#391 ).

Regards,
Kayman

ROBOTIS-Kayman, 2019-08-21:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-21 by ROBOTIS-Kayman