ROBOTIS-GIT/turtlebot3 #449: nav2_util build error in ros2 tutorial

eavelardev, 2019-07-09:
ros2 tutorial error
```
colcon build --symlink-install
```
--- stderr: nav2_util                                                                                                                                                               
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp: In function ‘rclcpp::Node::SharedPtr nav2_util::generate_internal_node(const string&)’:
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: error: ‘NodeOptions’ is not a member of ‘rclcpp’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:68:11: note: suggested alternative: ‘InitOptions’
   rclcpp::NodeOptions options;
           ^~~~~~~~~~~
           InitOptions
~/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/node_utils.cpp:69:3: error: ‘options’ was not declared in this scope
   options.use_global_arguments(false);
   ^~~~~~~
make[2]: *** [CMakeFiles/nav2_util_core.dir/src/node_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_util_core.dir/all] Error 2
make: *** [all] Error 2

######

I don't know if I need to install a dependency, the error don't mention it

routiful, 2019-07-15:
@Yeshasvitvs 
I've done to compile successfully. Did you try it with binary version of Dashing?

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460
I will comment about it later to check it.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

This issue will be solved due to navigation2 can be installed by binary.

closed 2019-08-21 by routiful