ROBOTIS-GIT/turtlebot3 #450: wrong moving behaviour of Turtlebot3

Adel-kn, 2019-07-16:
-Robot: waffle Pi
-SBC: Raspberry Pi 3
- OS installed in SBC: Raspbian
- OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)

Hello, 
i'm using Ubunto 16.04 to work on ROS kinetic version 1.
I try to apply a node to make the robot "turtlebot3" move "forward/backward" .
When I apply the node in the simulated environment "turtlebot3_fake" it works perfectly, but when i apply it on the robot, I got unexpected behaviour: e.g. when I give the command to robot to move forward only 1m then it keep moving  2.5m or 3m.
this problem happen also with rotational moving (rotating the robot around itself).
thanks in advance

rjshim, 2019-07-17:
@Adel-kn 

When you give forward/back commands to TB3, you are giving velocity commands (not joint position commands), which means that you are making it move forward/back at a certain speed, not making it move to a certain point.

On the other hand, if you use turtlbot3_fake, because the node makes the robot stop after a certain period of time (have a look at [turtlebot3_fake.cpp](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/master/turtlebot3_fake/src/turtlebot3_fake.cpp), Line 216 ), so it looks like you gave a command of how far the robot should move but what actually happened is that you gave a speed command and the robot stopped after cmd_vel_timeout_[s].

Let us know if you have any other questions, 
Ryan

Adel-kn, 2019-07-17:
Thank you Ryan for the reply. Tomorrow I will try to do something depending on your advice and hope it works. But how could one make the Robot drive to a certain position independently on the speed?

rjshim, 2019-07-18:
@Adel-kn 

Please refer to the following [demo](http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_examples/#point-operation). 

I recommend you go through the entire eManual first, explore functions provided, and then do your own project thing. 

Regards,
Ryan

Adel-kn, 2019-07-18:
@rjshim  Thanks Ryan

closed 2019-07-19 by rjshim