ROBOTIS-GIT/turtlebot3 #451: Max cmd_vel value

zimbot97, 2019-07-18:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [*] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [*] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [*] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - rostopic pub /cmd_vel geometry_msgs/Twist  '{linear:  {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}'
 
7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    -My turtlebot will not move when i publish that value. However, when I publish with X=0.1, it moved.

 What is the maximun cmd_vel value i can put for linear X motion?

ROBOTIS-Kayman, 2019-07-19:
It's 0.26 m/s.
You can find information about turtlebot3 in the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/specifications/#specifications).

Regards,
Kayman

zimbot97, 2019-07-19:
does this mean that when i publish cmd_vel, the max value is 0.26 for linear X? i cannot use values like 2.0, 3.0, etc?

ROBOTIS-Kayman, 2019-07-19:
No, you can not.
It's the maximum linear velocity of the turtlebot3, calculated as the maximum speed of the actuator.

closed 2019-07-19 by zimbot97