ROBOTIS-GIT/turtlebot3 #452: Problems with Dashing firmware

bengelman, 2019-07-19:
Hello,
About two months ago, I built a set of OpenEmbedded layers to build the Turtlebot3 demos with ROS Crystal. As Dashing has recently been released, over the last few days I have been trying to get the Turtlebot3 demos running with Dashing, and I've been encountering some issues with getting the firmware to work properly. I am using the latest Dashing releases of all relevant software and firmware. I've pieced together the workflow to the best of my ability, but it is more than likely that I may be missing a step, in which case please let me know. My process is as follows:

1. Cross-compile the latest version of Micro XRCE Agent and Client, and the Micro ROS Agent
2. Cross-compile the Turtlebot3 ROS nodes
3. (On the target) Copy DEFAULT_FASTRTPS_PROFILES.xml built by Micro ROS Agent, and tb3_fastrtps_profile.refs from tb3_sbc_settings, into the home directory
4. Modify tb3_fastrtps_profile.refs to reference the Turtlebot3 sound message instead of the absent builtin_interfaces sound message
5. Upload the opencr firmware, using the command "./update.sh /dev/ttyACM0 burger" 
(This step gives the same output as when I got the firmware working properly on Crystal)
6. In the directory with the FastRTPS profile files, execute "MicroXRCEAgent serial --dev /dev/ttyACM0 -r tb3_fastrtps_profile.refs"; this gives the following output:
info | SerialServerLinux.cpp | init | running... | fd: 3
7. Source the ROS setup script, then run "ros2 run turtlebot3_teleop teleop_keyboard". The MicroXRCEAgent window then displays:
info | Root.cpp | create_client | create | client_key: 0xAABBCCDD, session_id: 0x81
info | SerialServerBase.cpp | on_create_client | session established | client_key: 0xAABBCCDD, address: 0

All of this seems to indicate that initialization was successful; however, the robot will not move. It does not give any error messages, and no terminal output indicates that something is wrong, but the Turtlebot cannot be controlled by the keyboard. I have confirmed that the LIDAR works properly, so I can at least verify that the problem is probably not with the Micro XRCE Agent. Any assistance is appreciated, thanks!

ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [x] etc (PLEASE, WRITE DOWN YOUR OS)
OpenEmbedded Linux
4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

oneattosecond, 2019-07-19:
Try recompiling the OpenCR firmware via Arduino IDE & uploading via micro-USB from PC --> OpenCR. You might also try enabling DEBUG which pipes out OpenCR log messages via Serial port

You'll need ros2arduino library installed first via Tools-->Manage Libraries

https://github.com/ROBOTIS-GIT/ros2arduino
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#alternative-arduino-ide

mlherd, 2019-07-20:
@bengelman have you checked #426 and #442. I believe this is an ongoing issue. I had the same problem a few weeks ago when trying to use Turtlebot 3 with ROS2 Dashing.

mlherd, 2019-07-20:
@oneattosecond I will test Turtlebot 3 with ROS2 following your suggestions as well. Were you able to use Turtlebot 3 with ROS2 Dashing? Thanks.

bengelman, 2019-07-22:
@oneattosecond Thanks for your suggestion. I tried following the instructions from the links you provided, but it didn't seem to work (SW 1 and 2 did not cause the motors to move after uploading the firmware). In any case, I'd really like to get this working completely on the Pi without the aid of a remote PC, so I'll have to stick to the Crystal version until the Dashing firmware installer gets fixed.

@mlherd Thanks for your input. I had seen those issues, but there were replies suggesting that teleop should be working or that an updated Micro XRCE Agent might fix the problem, so I wasn't sure whether this was a universal issue. It's good to know that others are having this same problem.

mlherd, 2019-07-24:
@bengelman Thanks for your reply. I hope the firmware will support ROS2 Dashing distribution soon :)

OpusK, 2019-07-30:
Hi, all.

I have made an OpenCR firmware update, and have made an e-manual update.
I did not completely check everything I had to check, but I did update after the simple test.
Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2).

This is a temporary update, and since most developers are currently on vacation, it is likely that developers will be able to begin detailed checks next week.

And,
@bengelman 

> I tried following the instructions from the links you provided, but it didn't seem to work (SW 1 and 2 did not cause the motors to move after uploading the firmware). 

The TB3 binary file was fine, but this problem occurred when one of the contributors modified the code.
This issue is currently fixed in OpenCR 1.4.9. Please try after the update.

mlherd, 2019-07-30:
@OpusK Thanks for the update! I am going to try it as soon as possible.

mlherd, 2019-07-31:
@OpusK I had a chance to test it today. Turltlebot 3 works with ROS2 Dashing now. Thanks!

rafaelhuang-intel, 2019-07-31:
@mlherd 
@OpusK 

I also try it, but build fail at SBC setup (15.1.2) of Micro-XRCE-DDS-Agent Install Fail,  fail log attached, do you meet same problem ?

[tb3_for_dashing_build_fail.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/3450486/tb3_for_dashing_build_fail.txt)

bengelman, 2019-07-31:
@OpusK 
Thanks! It's now working for me. @rafaelhuang-intel it looks to me like your issue may be an Arm vs. Intel problem, and you might want to consider opening a new issue. At this point, I think the firmware issue is resolved, so I'm going to mark this as closed.

closed 2019-07-31 by bengelman