ROBOTIS-GIT/turtlebot3 #455: Installing turtlebot3 on Ubuntu IoT 18.04 architecture not detected properly

zmk5, 2019-07-25:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [x] Ubuntu 18.04 IoT

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - Going through the code provided [here](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#opencr-setup). I hit a snag running the following command:

```bash
~$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
```
 
7. Copy and Paste your error message on terminal

    - The above code results in this output:

```
aarch64
OpenCR Update Start..
./update.sh: line 27: ./opencr_ld_shell_x86: cannot execute binary file: Exec format error
```
  
8. Please, describe detailedly what difficulty you are in 

Installing on Ubuntu 18.04 IoT, the script attempts to install the x86 binaries even though we are on an arm system. Using the `uname -m` or `uname -p` returns:

```
aarch64
```

I believe a fix for this is to add this case to the `update.sh`:

```
case $(uname -m) in
    i386)   architecture="386" ;;
    i686)   architecture="386" ;;
    x86_64) architecture="amd64" ;;
    armv7l) architecture="arm" ;;
    aarch64) architecture="arm";;  # <-- my fix
    arm)    dpkg --print-architecture | grep -q "arm64" && architecture="arm64" || architecture="arm" ;;
esac
```

however, this runs into the following error:

```
-bash: ./opencr_ld_shell_arm: No such file or directory
```

I believe this is due to the binary file only compiled for a 32-bit arm architecture. Is it possible to get it for a 64-bit architecture?

zmk5, 2019-07-25:
After researching this more I realized that the update binary file `opencr_ld_shell_arm` is written for `arm7l` which is 32-bit, but the processor on both the Raspberry Pi 3 B and 3 B+ is 64-bit (Broadcom BCM2837).

Ubuntu Server IoT 18.04 uses `aarch64`, which is a 64-bit OS, while Ubuntu Mate 16.04+ still uses `arm7l`, which is 32-bit.

I checked using the `uname -a` command on both a turtlebot running Ubuntu Mate 16.04 and Ubuntu Server IoT 18.04, respectively.

**Ubuntu Mate 16.04**:

```bash
Linux turtlebot-00 4.4.38v7+ #938 SMP <DATE TIME ETC>  armv7l armv7l armv7l GNU/Linux
```

**Ubuntu Server IoT 18.04**:

```bash
Linux tb-01-pi 4.15.0-1031-raspi2 #33-Ubuntu SMP PREEMPT <DATE TIME ETC> aarch64 aarch64 aarch64 GNU/Linux
```

OpusK, 2019-07-29:
Hi, @zmk5 

As you guessed, excutable is compiled with 32bit. Therefore, 32bit support package must be installed for 64bit operating system. (OS Dependency)

Unfortunately, currently I have no environment to test aarch64 hardware / OS.
So, please refer to the link below.

- [32 Bit executables in AARCH64 system](https://forum.odroid.com/viewtopic.php?t=18806)

zmk5, 2019-07-29:
Hi!

I was able to install the update using a copy of Ubuntu Mate `armhf` microSD card I had. Just swapped out the SD cards installed, and put back 18.04.

> Unfortunately, currently I have no environment to test aarch64 hardware / OS.
So, please refer to the link below.

I believe the processor on the Raspberry Pi 3 is already 64-bit, its just the OS that is the issue. I'll test the software and let you know how to works with `aarch64` OS. :-)

OpusK, 2019-07-30:
@zmk5 

> I believe the processor on the Raspberry Pi 3 is already 64-bit

Yes. The hardware supports 64-bit.
I currently have no hardware to test. I will get the hardware out next week and I will test it.
In addition, please refer to the link below.
 - [Running 32 bit applications on Aarch64](https://forum.armbian.com/topic/4764-running-32-bit-applications-on-aarch64/)

robotpilot, 2019-08-01:
- Related post:
ROS Discourse
https://discourse.ros.org/t/setting-up-the-turtlebot3-with-ros2-on-ubuntu-server-iot-18-04/10090
- Related issue:
TurtleBot3 https://github.com/ROBOTIS-GIT/turtlebot3/issues/460

zmk5, 2019-08-06:
Hey @robotpilot! I actually wrote that blog post. I updated it as I was doing some of the stuff for this post :-)

OpusK, 2019-08-12:
Hi, @zmk5 

Please try again after the command below.
I will update the contents of update.sh.

```bash
sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf
```

OpusK, 2019-08-12:
@zmk5, @robotpilot 

I have updated the update.sh file. 
Please use it with the following command:
- [TB3 update archives](https://github.com/ROBOTIS-GIT/OpenCR-Binaries/tree/master/turtlebot3/ROS2/latest)

```bash
sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf
```

routiful, 2019-08-21:
@zmk5 

The issue seems to be solved. If you have any question about this issue, please feel free to reopen or get a new ticket :)

closed 2019-08-21 by routiful

zentala, 2020-03-26:
It doesn't help me. 

```
ubuntu@ubuntu:~$ export OPENCR_PORT=/dev/ttyACM0
ubuntu@ubuntu:~$ export OPENCR_MODEL=waffle
ubuntu@ubuntu:~$ rm -rf ./opencr_update.tar.bz2
ubuntu@ubuntu:~$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
--2020-03-26 23:40:35--  https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 [following]
--2020-03-26 23:40:35--  https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR-Binaries/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 462386 (452K) [application/octet-stream]
Saving to: ‘opencr_update.tar.bz2’

opencr_update.tar.bz2         100%[==============================================>] 451.55K  1.73MB/s    in 0.3s    

2020-03-26 23:40:35 (1.73 MB/s) - ‘opencr_update.tar.bz2’ saved [462386/462386]

opencr_update/burger_turtlebot3_core.ino.bin
opencr_update/om_with_tb3.opencr
opencr_update/waffle_turtlebot3_core.ino.bin
opencr_update/opencr_ld_shell_arm
opencr_update/update.sh
opencr_update/
opencr_update/released_1.2.5.txt
opencr_update/waffle.opencr
opencr_update/opencr_ld_shell_x86
opencr_update/burger.opencr
opencr_update/open_manipulator_turtlebot3_core.ino.bin
aarch64

OpenCR Update Start..
./update.sh: line 27: ./opencr_ld_shell_x86: cannot execute binary file: Exec format error
ubuntu@ubuntu:~/opencr_update$ ls /dev/
Display all 934 possibilities? (y or n)
ubuntu@ubuntu:~/opencr_update$ ls /dev/ttyACM0 
/dev/ttyACM0
ubuntu@ubuntu:~/opencr_update$ sudo dpkg --add-architecture armhf
ubuntu@ubuntu:~/opencr_update$ sudo apt-get update
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease                
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease              
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Hit:5 http://packages.ros.org/ros/ubuntu bionic InRelease                    
Hit:6 http://packages.ros.org/ros2/ubuntu bionic InRelease
Reading package lists... Done
ubuntu@ubuntu:~/opencr_update$ sudo apt-get install libc6:armhf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libc6:armhf is already the newest version (2.27-3ubuntu1).
The following package was automatically installed and is no longer required:
  linux-raspi2-5.3-headers-5.3.0-1017
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 99 not upgraded.
ubuntu@ubuntu:~/opencr_update$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
aarch64

OpenCR Update Start..
./update.sh: line 27: ./opencr_ld_shell_x86: cannot execute binary file: Exec format error
ubuntu@ubuntu:~/opencr_update$ 
```

Ubuntu 18.04 on RPi4.

thillRobot, 2020-11-10:
Hello. I am running into this smae issue now when following the tutorial here ([https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup]). I am running a pi3B+ and Mate18.04 arm64. From what I have read in this post, it should work. However, I receive the error mentioned above when attempting **6.3.1.1 - Recommended Shell Script** in the tutorial. 

```
OpenCR Update Start..
./update.sh: line 27: ./opencr_ld_shell_x86: cannot execute binary file: Exec format error

```

Also, I looked in `update.sh` and the edit you mentioned is not there. I made the edit and ran the following three lines (which are in the tutorial). 

```
sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf
```

I am still seeing the same error. Should I try the 32bit image instead, or is there a known fix for this issue?

zmk5, 2020-11-11:
@thillRobot not sure why your installation still assumes you are running an x86. What do you get when you type the command `uname -m` into the terminal?

thillRobot, 2020-11-11:
I receive `aarch64` after entering `uname -m` into the terminal.

zmk5, 2020-11-11:
try going into the `update.sh` file and add the following line:

```
case $(uname -m) in
    i386)   architecture="386" ;;
    i686)   architecture="386" ;;
    x86_64) architecture="amd64" ;;
    armv7l) architecture="arm" ;;
    aarch64) architecture="arm";;  # <-- ADD THIS LINE.
    arm)    dpkg --print-architecture | grep -q "arm64" && architecture="arm64" || architecture="arm" ;;
esac
```

Then try running it again.

thillRobot, 2020-11-11:
Great. That is the fix! I tried that last night, but I was not paying close enough attention. I re-ran the whole line with the `wget` and all after a making that edit.  

Thanks for your help!

zmk5, 2020-11-11:
Of course habibi! Glad to help!