ROBOTIS-GIT/turtlebot3 #457: Raspi

previns, 2019-07-26:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    **- [X ] Burger**
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    **- [X ] Raspberry Pi 3**
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    **- [ X] Ubuntu MATE 16.04 or later**
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    **- [X ] Ubuntu 16.04 LTS (Xenial Xerus)**
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [16.04.6]
    - Firmware version: [x.x.x] (NOT SURE?)
 
6. Write down the commands you used in order
      **[IP ADDRESSES]**
REMOTE PC: 10.0.0.4
TURTLEBOT1: 10.0.0.151
TURTLEBOT2: 10.0.0.169
MASTER_URI: REMOTE PC (10.0.0.4:11311)

    **[REMOTE PC]**
`roscore`

    **[TURTLEBOT1]**
 `roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch`

    **[REMOTE PC]**
`rqt_image_view`
Doesnt show any camera input (shows some gradient)

    **[TURTLEBOT2]**
 `roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch`

    **[REMOTE PC]**
`rqt_image_view`
Doesnt show any camera input (shows some gradient)

7. Copy and Paste your error message on terminal

    **[TURTLEBOT1]**
  `
... logging to /home/turtlebot/.ros/log/91d93322-af77-11e9-9ca3-0800274a06c7/roslaunch-turtlebot-desktop-2904.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.151:36869/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: camera
 * /raspicam_node/camera_info_url: package://turtleb...
 * /raspicam_node/framerate: 15
 * /raspicam_node/height: 480
 * /raspicam_node/width: 640
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://10.0.0.4:11311

process[raspicam_node-1]: started with pid [2914]
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.ril.camera' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (7)
mmal: Failed to create camera component
[ERROR] [1564130317.623447625]: Failed to create camera component
[ERROR] [1564130317.623755020]: init_cam: Failed to create camera component
[ INFO] [1564130317.623992416]: camera calibration URL: package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
[ WARN] [1564130317.681647491]: [camerav2_1280x960] does not match name camera in file /home/turtlebot/catkin_ws/src/turtlebot3/turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
[ INFO] [1564130317.681826605]: Camera successfully calibrated from default file
[ INFO] [1564130317.681906813]: using default calibration URL
[ INFO] [1564130317.682033845]: camera calibration URL: file:///home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml
[ INFO] [1564130317.682231345]: Unable to open camera calibration file [/home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml]
[ WARN] [1564130317.682318844]: Camera calibration file /home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml not found.
[ INFO] [1564130317.682410251]: No device specifc calibration found
[ WARN] [1564130318.179409734]: reconfiguring, but camera_component not initialized
[FATAL] [1564130318.386946971]: Tried to start capture before camera is inited
[raspicam_node-1] process has died [pid 2914, exit code -11, cmd /home/turtlebot/catkin_ws/devel/lib/raspicam_node/raspicam_node __name:=raspicam_node __log:=/home/turtlebot/.ros/log/91d93322-af77-11e9-9ca3-0800274a06c7/raspicam_node-1.log].
log file: /home/turtlebot/.ros/log/91d93322-af77-11e9-9ca3-0800274a06c7/raspicam_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
turtlebot@turtlebot-desktop:~$
`
    **[TURTLEBOT2]**
... logging to /home/turtlebot/.ros/log/91d93322-af77-11e9-9ca3-0800274a06c7/roslaunch-turtlebot-desktop-3890.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.169:40945/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: camera
 * /raspicam_node/camera_info_url: package://turtleb...
 * /raspicam_node/framerate: 15
 * /raspicam_node/height: 480
 * /raspicam_node/width: 640
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://10.0.0.4:11311

process[raspicam_node-1]: started with pid [3899]
[ INFO] [1564130121.895724136]: splitter component done

[ INFO] [1564130122.009768738]: camera calibration URL: package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
[ WARN] [1564130122.102275887]: [camerav2_1280x960] does not match name camera in file /home/turtlebot/catkin_ws/src/turtlebot3/turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
[ INFO] [1564130122.102475259]: Camera successfully calibrated from default file
[ INFO] [1564130122.102578121]: using default calibration URL
[ INFO] [1564130122.102697234]: camera calibration URL: file:///home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml
[ INFO] [1564130122.102892438]: Unable to open camera calibration file [/home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml]
[ WARN] [1564130122.102990509]: Camera calibration file /home/turtlebot/.ros/camera_info/camerav2_1280x960.yaml not found.
[ INFO] [1564130122.103074518]: No device specifc calibration found
[ INFO] [1564130122.883528393]: Starting video capture (640, 480, 80, 15)

[ INFO] [1564130122.884109111]: Video capture started

^C[raspicam_node-1] killing on exit
raspicam_node: /usr/include/boost/thread/pthread/recursive_mutex.hpp:113: void boost::recursive_mutex::lock(): Assertion `!pthread_mutex_lock(&m)' failed.
shutting down processing monitor...
... shutting down processing monitor complete
done
turtlebot@turtlebot-desktop:~$

8. Please, describe detailedly what difficulty you are in 

   I have 2 Turtlebot3 Burger and I want to make my senior design project where I control one turtlebot with bluetooth controller (probably playstation 4 or xbox one controller) and the other turtlebot will follow. I want to code this using Matlab, however, currently I am just trying to start communication between matlab and one turtlebot.

When I perform` ROSTOPIC LIST` on Matlab, I do not see `/raspicam_node/image/compressed `or `raspi_cam` or anything else to do with that camera.

Even when I launch

`roslaunch turtlebot3_bringup turtlebot3_robot.launch `

on turtlebot, I only get the following topics:

`>> rostopic list
/battery_state                       
/cmd_vel                             
/cmd_vel_rc100                       
/diagnostics                         
/firmware_version                    
/imu                                 
/joint_states                        
/magnetic_field                      
/motor_power                         
/odom                                
/raspicam_node/camera_info           
/raspicam_node/image/compressed      
/raspicam_node/parameter_descriptions
/raspicam_node/parameter_updates     
/reset                               
/rosout                              
/rosout_agg                          
/rpms                                
/scan                                
/scan_filtered                       
/sensor_state                        
/sound                               
/tf                                  
/version_info `

I want to figure out how to move turtlebot, use camera, and use lidar.

The only thing I have working so far is Lidar.

Not sure what is wrong with camera, but I cannot find raspi camera topics, and I also cannot move turtlebot? Please advise.

rjshim, 2019-07-29:
Hi @previns 

You will need to refer to [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/teleoperation/#ps3-joystick) and [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#turtlebot-follower-demo), and you will also need to port them to Matlab on your own if needed since we do not officially support Matlab.

If you have questions about ROS Matlab, please ask them at this [link](https://kr.mathworks.com/help/robotics/examples/get-started-with-ros.html)

Regards,
Ryan

rjshim, 2019-09-26:
This issue will be closed since there has been no response for a while. You can reopen this issue if you have any relevant questions. Thanks.

Ryan

closed 2019-09-26 by rjshim