ROBOTIS-GIT/turtlebot3 #458: IMU timestamps go backwards with dashing (1.0.2 OpenCR) - crashes cartographer

oneattosecond, 2019-07-30:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3

3. Which OS you installed in SBC?

    - [ ] Raspbian

4. Which OS you installed in Remote PC?

    - [ ] OS X

5. Write down software version and firmware version

    - Software version: [???]
    - Firmware version: [1.0.2]. -- firmware from here: https://github.com/ROBOTIS-GIT/OpenCR-Binaries/blob/master/turtlebot3/ROS2/1.0.2/opencr_update_1.0.2.tar.bz2

This contains the workaround from https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/99
 
6. Write down the commands you used in order

    - Raspberry pi: run MicroXRCE agent for serial & udp, run turtlebot_lidar 
    - remote PC: ros2 topic echo imu | tee imu_data_jul30.txt'

    TB3 waffle pi is sitting idle no movement and still timestamp (nanoseconds) is rolling backwards...
 
7. Copy and Paste your error message on terminal

    - There is no error message on terminal, but IMU data has timestamps going backwards which causes multiple crashes in cartographer
  
8. Please, describe detailedly what difficulty you are in 

    - cartographer crashes all the time and is totally unusable due to hitting asserts in code because time is going backwards

Example IMU timestamp going backwards:

---
header:
  stamp:
    sec: 1564510494
    nanosec: 375714706   <-- nanoseconds here
  frame_id: imu_link
orientation:
  x: 0.013068560510873795
...
---
header:
  stamp:
    sec: 1564510494
    nanosec: 15053893       <-- nanoseconds here is smaller than before...
  frame_id: imu_link
orientation:
  x: 0.011846276931464672
...

This causes a number of asserts in cartographer to be hit and cannot complete mapping process.

routiful, 2019-08-01:
We opened new issue page to summary every issues related turtlebot3 ROS2 #460 
I will comment about it later to check it.

OpusK, 2019-08-05:
Hi, @oneattosecond 

This is probably due to network environment/performance and QoS settings.
This is an issue that occurs when ROS2 uses DDS, and you can add/modify QoS by modifying the `~/tb3_sbc_settings/tb3_fastrtps_protile.refs` file.

My guess is that now KEEP_LAST is set to 10, so the previous data is coming up later.
Therefore, please change this to 1 and test it.

Please refer to the eProsima document below for QoS settings.
- [About XML Representation on Micro XRCE-DDS Client](https://micro-xrce-dds.readthedocs.io/en/latest/operations.html#about-xml-representation)
- [XML profiles of Fast-RTPS](https://eprosima-fast-rtps.readthedocs.io/en/latest/xmlprofiles.html#xml-profiles)

oneattosecond, 2019-08-05:
@OpusK I compiled Micro-XRCE from source on top of stock Raspbian Buster, so I do not have tb3_fastrtps_protile.refs and cannot try your suggestion. What is the default QOS profile for Micro-XRCE DDS client? It would seem I am using the default, as I do not have tb3_fastrtps_protile.refs

However, I do not think your suggestion will help. The timestamp is created on OpenCR side, in function publishImu() which calls:

    msg->header.stamp    = ros2::now();

Unless MicroXRCE is changing the timestamp in msg->header.stamp, your suggestion cannot help, and the root cause is more likely the timestamp coming from ros2::now() is broken.

Can you please investigate further?

Regards

oneattosecond, 2019-08-05:
~/Micro-XRCE-DDS-Agent/test/agent.refs

This file had a KEEP_LAST value set to 10, I changed it to 1, and still saw timestamps go backwards.

OpusK, 2019-08-06:
@oneattosecond 

>  I compiled Micro-XRCE from source on top of stock Raspbian Buster, so I do not have tb3_fastrtps_protile.refs and cannot try your suggestion. 

It seems to me that you did not proceed with the installation according to the e-manual. Is that correct?

Please refer to the e-manual installation instructions and try again.
 - [TB3 ros2](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

oneattosecond, 2019-08-06:
Yes, I followed the wiki quite strictly. However there are as you know, many steps. I followed all the steps in install.sh, 

Can you confirm you never see timestamps go backwards? I am just doing "ros2 topic echo imu" for 10 minutes and observing timestamp goes backwards.

Do you plan to publish a Raspberry Pi .img for Dashing? That would help reduce possible errors.

OpusK, 2019-08-06:
@oneattosecond 

> Yes, I followed the wiki quite strictly. However there are as you know, many steps. I followed all the steps in install.sh,

If you followed the instructions in the e-manual, the `tb3_fastrtps_profile.refs` file must exist.
In e-manual, the following compressed file is downloaded. In this compressed file, the `tb3_fastrtps_profile.refs` file exists.
 - [tb3_sbc_settings.tar.bz2](https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_sbc_settings/latest/tb3_sbc_settings.tar.bz2)

> Can you confirm you never see timestamps go backwards?

By the time I tested, I haven't found any problems yet.

> Do you plan to publish a Raspberry Pi .img for Dashing? That would help reduce possible errors.

As @routiful mentioned(#460), we are planning to apply other communication methods than XRCE. After this, we will consider it with the team.

oneattosecond, 2019-08-06:
OK, I have deleted everything I did before and run install.sh, double check the wiki for any missing steps (none found). Then I modified the KEEP_ALIVE to 1 in tb3_fastrtps_profile.refs and am running all the software as expected via run.sh

I double checked and my OpenCR is running 1.0.2 which has the workaround for eProsima issue.

I am still seeing many timestamps go backwards over a period of 10 minutes... I wrote a quick script to verify, and the timestamp went backwards (nanoseconds_cur < nanosec_prev within same second) 30 times in 10 minutes.

Can you please investigate? I am suspecting the below code is giving wrong timestamp:

   msg->header.stamp    = ros2::now();

OpusK, 2019-08-07:
@oneattosecond 

> I am still seeing many timestamps go backwards over a period of 10 minutes... I wrote a quick script to verify, and the timestamp went backwards (nanoseconds_cur < nanosec_prev within same second) 30 times in 10 minutes.
Can you please investigate? I am suspecting the below code is giving wrong timestamp:
msg->header.stamp = ros2::now();

I've never seen this, but I'll check the code anyway

Note, as shown in #460, we are trying to apply a different communication method to TB3. In that case, TB3 will not use ros2arduino.
So, after this development, I'll check the issue with `ros2::now()`.

routiful, 2019-08-21:
For now on, we have been released TB3 ROS 2 Dashing!
Please refer to [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

IMU timestamps issues will be solved to download new firmware.

closed 2019-08-21 by routiful