ROBOTIS-GIT/turtlebot3 #460: ROS 2 Dashing Diademata

robotpilot, 2019-08-01:
This meta ticket aggregates the tasks targeted for the TurtleBot3 and ROS 2 Dashing Diademata. Release date is August 20, 2019. Tasks will be updated as they progress.

1. Porting to `ROS 2 Dashing Diademata`
    - [x] turtlebot3
    - [x] turtlebot3_msgs
    - [x] turtlebot3_simulations
    - [ ] turtlebot3_applications_msgs
    - [ ] turtlebot3_applications
    - [x] hls_lfcd_lds_driver
    - [ ] Documentation

2. Support `ros2arduino` and `Micro XRCE-DDS Agent`
: This method uses [DDS-XRCE](https://www.omg.org/spec/DDS-XRCE/About-DDS-XRCE/) to communicate between `OpenCR` and `Raspberry Pi` in TurtleBot3. The `ros2arduino` is the ROS 2 package for `Arduino` developed by `ROBOTIS` and The `Micro XRCE-DDS Agent` is the agent program for `DDS-XRCE` developed by `eProsima`.
    - [x] [ros2arduino](https://github.com/ROBOTIS-GIT/ros2arduino)
    - [x] [Micro XRCE-DDS Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent) by eProsima
    - [ ] Documentation

3. Support `Dynamixel Protocol` and `Raspbian Image`
: This method does not use the `DDS-XRCE` described above. It communicates with `OpenCR` and `Raspberry Pi` using [Dynamixel Protocol](http://emanual.robotis.com/docs/en/dxl/protocol2/) which is a specific serial protocol. This method provides a Raspbian image with pre-installed ROS 2 Dashing.
    - [x] [Dynamixel Protocol 2.0](http://emanual.robotis.com/docs/en/dxl/protocol2/)
    - [ ] OpenCR firmware for this method
    - [ ] Raspbian image with pre-installed ROS 2 Dashing
    - [ ] Documentation

rafaelhuang-intel, 2019-08-01:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/461

rafaelhuang-intel, 2019-08-02:
report new SLAM simulate issue(15.6)  https://github.com/ROBOTIS-GIT/turtlebot3/issues/463

OpusK, 2019-08-05:
@rafaelhuang-intel  

As I replied to #461, this is an issue related to eProsima's agent installation/uninstallation.

oneattosecond, 2019-08-13:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/469

Can someone else confirm cartographer works for them on latest sources? I cannot get scan or scan_half topics due to above.

mlherd, 2019-08-21:
@oneattosecond I tested it last week. I was able to create maps using cartographer. It wasn't perfect, but it worked. Also, I was navigate in the created map using Navigation 2.

[Cartographer Video](https://drive.google.com/open?id=1skUR8d9to8gYEwwuIhLYol-ii6X5PjPd)

[Navigation 2 Video](https://drive.google.com/open?id=1kNziYLY-mO-Nh_b0Pmypla58od9BhkqG)

routiful, 2019-08-21:
TurtleBot3 ROS 2 Dashing version was released :)

Please refer to this [link](https://discourse.ros.org/t/tb3-ros-2-dashing-release/10364) and [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

rafaelhuang-intel, 2019-08-21:
Hi, @routiful 
 
I've a question, I no Raspberry Pi 3, so I want to install all package & component on same PC (NUC), so it's means I only need skip 15.1.2.1 & 15.1.2.2?

routiful, 2019-08-21:
@rafaelhuang-intel 
Hi :)
Of cource, you don't need to follow 15.1.2.1 and 15.1.2.2.

yoshimalucky, 2019-08-21:
Now, I'm working on:
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#install-turtlebot3-packages

However, the system is hung up with the command:
colcon build --symlink-install --parallel-workers 1

Build process of turtlebot3_node stops at 28%:

Starting >>> turtlebot3_node
[25.3s] [4/8 complete] [turtlebot3_node:build 28% - 13.3s]
[Processing: turtlebot3_node]
[Processing: turtlebot3_node]
[3min 31.5s] [4/8 complete] [turtlebot3_node:build 28% - 3min 19.4s]

Also, another terminal screen says:
INFO: task kworker/1:1:32 blocked for more than 120 seconds.
Not tainted 4.15.0-1043-raspi2 #46-Ubuntu
"echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disable this message.

routiful, 2019-08-21:
@yoshimalucky 
Would you try to increase swap memory on raspi3?
https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/

yoshimalucky, 2019-08-21:
@routiful Good hint! Thanks!

The instruction is for rasbian, so it does not work for our case because our emanual tell us to use ubuntu server image.

Therefore, I followed this instruction to increase swap on a ubuntu system:
https://linuxize.com/post/how-to-add-swap-space-on-ubuntu-18-04/

Then, I could build them from the source! Thanks!

routiful, 2019-08-21:
@yoshimalucky 
Nice :) I will be update this issue on e-manual as soon as possible. Thanks!

rafaelhuang-intel, 2019-08-22:
Hi, @routiful 

I install dashing release all package & component on same PC (NUC),  I skip 15.1.2.x all step, it's workable. 

I tried SLAM & navigation 2  both on real TB3 & gazebo, only SLAM simulate issue still exist (https://github.com/ROBOTIS-GIT/turtlebot3/issues/463).

rafaelhuang-intel, 2019-08-22:
@routiful 

When I install dashing release, I meet opencr update need /dev/ttyACM0 need permission and I set user to dialout to fix this problem. I hope to know should turtlebot3_bringup need this permission? if yes, suggestion you can add comments in e-manual.

routiful, 2019-08-22:
@rafaelhuang-intel 

Sorry for inconvinience. I updated e-manual to solve releated issue.
Please follow below link.
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#opencr-port-set-up

robotpilot, 2019-08-27:
- Known Issues with RViz2 and OpenSplice
: https://github.com/ros2/rviz/issues/437
- Known Issues with RViz2 and TF
: https://github.com/ros2/rviz/issues/359
: https://github.com/ros2/rviz/issues/332
- Known Issues with cartographer_ros package
: https://github.com/ros2/cartographer_ros/issues/33

closed 2020-03-12 by robotpilot

ioannagiorgi, 2023-07-17:
> Now, I'm working on: http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#install-turtlebot3-packages
>
> However, the system is hung up with the command: colcon build --symlink-install [...]

Hi, I am having the same issue, how did you solve this?

rafaelhuang-intel, 2023-07-18:
> > Now, I'm working on: http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#install-turtlebot3-packages
> > However, the system is hung up with the command: colcon build --symlink-install --parallel-workers 1
> > Build process of turtlebot3_node stops at 28%:
> > Starting >>> turtlebot3_node [25.3s] [4/8 complete] [turtlebot3_node:build 28% - 13.3s] [Processing: turtlebot3_node] [Processing: turtlebot3_node] [3min 31.5s] [4/8 complete] [turtlebot3_node:build 28% - 3min 19.4s]
> > Also, another terminal screen says: INFO: task kworker/1:1:32 blocked for more than 120 seconds. Not tainted 4.15.0-1043-raspi2 #46-Ubuntu "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disable this message.
> 
> Hi, I was having the same issue with Humble, but after setting the swap I was able to build it successfully.

if your PC performance is poor, suggest you remove build parameter "--parallel-workers 1" and run "colcon build --symlink-install"