ROBOTIS-GIT/turtlebot3 #461: TB3 build fail at SBC setup (15.1.2) of Micro-XRCE-DDS-Agent Install Fail

rafaelhuang-intel, 2019-08-01:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    -  Waffle

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    -  NUC6i7KYK

3. Which OS you installed in SBC?

    -  Ubuntu 18.04 LTS (Bionic Beaver)
    -  ROS2 Dashing

4. Which OS you installed in Remote PC?

    - Ubuntu 18.04 LTS (Bionic Beaver)
    -  ROS2 Dashing

5. Write down software version and firmware version

    - Firmware version: V190730R1
 

after follow http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2 to build TB3 for ROS2 dashing release, build fail at SBC setup (15.1.2) of Micro-XRCE-DDS-Agent Install Fail, fail log attached at below.

[tb3_for_dashing_build_fail.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/3454624/tb3_for_dashing_build_fail.txt)

rafaelhuang-intel, 2019-08-02:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/460

OpusK, 2019-08-05:
Hi, @rafaelhuang-intel 

This issue is probably caused by a conflict with the previous agent.
You can see this issue below and manually delete the existing agent's library file.
 - [How to uninstall Agent and Client](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/95)

rafaelhuang-intel, 2019-08-06:
Hi, @OpusK 

I've build a clean environment and test it.

First round is ok after build successful, but second round is fail while use 'ps | grep "MicroXRCEAgent"' command for checking process ID and kill it. and check topic list we can see lost more.

**Terminal 1**
intel@intel-desktop:~/tb3_sbc_settings$ cd ~/tb3_sbc_settings && ./run.sh

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

 use 'ps | grep "MicroXRCEAgent"' command for checking process ID
 kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
==========================

intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
intel@intel-desktop:~/tb3_sbc_settings$ ^C
intel@intel-desktop:~/tb3_sbc_settings$ ^C
intel@intel-desktop:~/tb3_sbc_settings$ ^C
intel@intel-desktop:~/tb3_sbc_settings$ ^C
intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
intel@intel-desktop:~/tb3_sbc_settings$ cd ~/tb3_sbc_settings && ./run.sh

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: nohup: ignoring input and appending output to 'nohup.out'
appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

 use 'ps | grep "MicroXRCEAgent"' command for checking process ID
 kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
==========================

intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"

**Terminal2**
intel@intel-desktop:~$ ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /home/intel/.ros/log/2019-08-06-10-43-34-936513-intel-desktop-27910
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
[INFO] [robot_state_publisher-1]: process started with pid [27921]
[INFO] [turtlebot3_ros-2]: process started with pid [27922]
[robot_state_publisher-1] Initialize urdf model from file: /home/intel/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 7 children
[... 25 lines cut ...]
[robot_state_publisher-1] Adding fixed segment from base_link to camera_link
[robot_state_publisher-1] Adding fixed segment from camera_link to camera_depth_frame
[robot_state_publisher-1] Adding fixed segment from camera_depth_frame to camera_depth_optical_frame
[robot_state_publisher-1] Adding fixed segment from camera_link to camera_rgb_frame
[robot_state_publisher-1] Adding fixed segment from camera_rgb_frame to camera_rgb_optical_frame
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_left_link
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_right_link
[robot_state_publisher-1] Adding fixed segment from base_link to imu_link
[robot_state_publisher-1] Adding fixed segment from base_link to base_scan
[robot_state_publisher-1] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-1] Adding moving segment from base_link to wheel_right_link

**Terminal3**
/imu
/initialpose
/joint_states
/move_base_simple/goal
/odom
/parameter_events
/robot_description
/rosout
/sensor_state
/tf
/tf_static
/time_sync

OpusK, 2019-08-06:
@rafaelhuang-intel 

> use 'ps | grep "MicroXRCEAgent"' command for checking process ID
kill [PID number]

You must check the process ID (PID) with ps and kill the process with the kill command.

rafaelhuang-intel, 2019-08-06:
@OpusK 

I tried reboot system & reset OpenCR, and than run launch command again, but still no PID output.

intel@intel-desktop:~/tb3_sbc_settings$ cd ~/tb3_sbc_settings && ./run.sh

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

 use 'ps | grep "MicroXRCEAgent"' command for checking process ID
 kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
==========================

intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
intel@intel-desktop:~/tb3_sbc_settings$

OpusK, 2019-08-06:
@rafaelhuang-intel 

Except for the `grep` command, can you retry using only the` ps` command?

rafaelhuang-intel, 2019-08-06:
@OpusK 

Except for the grep command,  and only run the ps command., no PID display. run lsusb can see the device already connected.

intel@intel-desktop:~/tb3_sbc_settings$ cd ~/tb3_sbc_settings && ./run.sh

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'
nohup: ignoring input and appending output to 'nohup.out'

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

 use 'ps | grep "MicroXRCEAgent"' command for checking process ID
 kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
==========================

intel@intel-desktop:~/tb3_sbc_settings$ ps
  PID TTY          TIME CMD
28487 pts/0    00:00:00 bash
29019 pts/0    00:00:00 ps
intel@intel-desktop:~/tb3_sbc_settings$ lsusb
Bus 002 Device 005: ID 05e3:0617 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 020: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 021: ID 0483:5740 STMicroelectronics STM32F407
Bus 001 Device 018: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 005: ID 413c:3012 Dell Computer Corp. Optical Wheel Mouse
Bus 001 Device 004: ID 413c:2112 Dell Computer Corp. 
Bus 001 Device 002: ID 14cd:8601 Super Top 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
intel@intel-desktop:~/tb3_sbc_settings$

OpusK, 2019-08-06:
@rafaelhuang-intel 

What operating system does your development environment use on x86?
This seems to be an issue between nohup, ps, and the OS.

I recommend you to run the commands in the `run.sh` file separately from each terminal.

rafaelhuang-intel, 2019-08-06:
@OpusK 
I installed Ubuntu 18.04 on Intel NUC(i7-6700HQ)

after run above 3 command, all fail and display libfastrtps.so.1 missing.

Terminal 1 (After reset OpenCR)
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 1000000 -r tb3_fastrtps_profile.refs -v 5
MicroXRCEAgent: error while loading shared libraries: libfastrtps.so.1: cannot open shared object file: No such file or directory

Terminal 2
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent udp -p 2019 -r tb3_fastrtps_profile.refs -v 5
MicroXRCEAgent: error while loading shared libraries: libfastrtps.so.1: cannot open shared object file: No such file or directory

Terminal 3
intel@intel-desktop:~/tb3_sbc_settings$ ./turtlebot3_lidar_xrce_client
Error at create session.

OpusK, 2019-08-06:
@rafaelhuang-intel 

> intel@intel-desktop

Are you using NUC instead of TB3's Raspberry Pi?

> MicroXRCEAgent: error while loading shared libraries: libfastrtps.so.1: cannot open shared object file: No such file or directory

It looks like the MicroXRCEAgent is not installed properly.
You need to verify that the agent is installed correctly.

rafaelhuang-intel, 2019-08-06:
@OpusK 

yes, using NUC instead of TB3's Raspberry Pi.

I tried first round workable, and follow e-manual (http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#setup) to run 15.2 ~15.3 are workable.
and than, I kill PID and want  to try run 15.4 (cartographer), I meet above issue.
so you means I need re-install 15.1.2(SBC setup) again?

ksd2201, 2019-08-06:
@OpusK 
hi i have same problem(install fail MicroXRCE)
you said this problem will occured relate with old MicroXRCE version
but i just install new RaspbianOS on claer SDcard.
and i don't understand this link https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/95
can you explain detail usage? 
thanks

rafaelhuang-intel, 2019-08-06:
@OpusK 

after re-install SBC setup, build issue exist, so can you explain detail usage of https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/95 ?

/usr/bin/ld: /usr/local/lib/libmicroxrcedds_client.a(session.c.o): relocation R_X86_64_PC32 against symbol `uxr_flash_output_streams' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
CMakeFiles/microxrcedds_agent.dir/build.make:1010: recipe for target 'libmicroxrcedds_agent.so.1.1.0' failed
make[5]: *** [libmicroxrcedds_agent.so.1.1.0] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/microxrcedds_agent.dir/all' failed
make[4]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/uagent.dir/build.make:112: recipe for target 'uagent-prefix/src/uagent-stamp/uagent-build' failed
make[2]: *** [uagent-prefix/src/uagent-stamp/uagent-build] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/uagent.dir/all' failed
make[1]: *** [CMakeFiles/uagent.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

 [Fail] Micro-XRCE-DDS-Agent Install Fail

OpusK, 2019-08-07:
@ksd2201, @rafaelhuang-intel 

I don't want to recommend eProsima/Micro-XRCE-DDS-Agent#95's way. This is because you need to link the library folder every time.

> but i just install new RaspbianOS on claer SDcard.

@ksd2201, This is a phenomenon I have never seen.
It is a good idea to raise an issue with eProsima because this is a problem with the Agent installation.

@rafaelhuang-intel, What files did you delete to remove Agent's previous installation library?

rafaelhuang-intel, 2019-08-07:
@OpusK 

only remove tb3_sbc_settings folder and unzip tar package again, so could tell me how many library need remove?

OpusK, 2019-08-07:
@rafaelhuang-intel 

You do not need to delete the tb3_sbc_settings folder. This is a problem only with Agent. You must delete all library files in the root directory.

Delete the library files that contain the word `microxrcedds` with the command below. (only for client and agent)

```bash
$ sudo find / "*microxrcedds*"
```

rafaelhuang-intel, 2019-08-07:
@OpusK 

The command of your provide not search any one, so I changed command to "sudo find / -name "*microxrcedds*"", and than all relate file are list in terminal, could you help to check if all file need delete?

intel@intel-desktop:~$ sudo find / -name "*microxrcedds*"
find: ‘/proc/1842/task/1842/net’: Invalid argument
find: ‘/proc/1842/net’: Invalid argument
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/CMakeFiles/microxrcedds_agent.dir
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/CMakeFiles/Export/share/microxrcedds_agent
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/CMakeFiles/Export/share/microxrcedds_agent/cmake/microxrcedds_agentTargets.cmake
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/CMakeFiles/Export/share/microxrcedds_agent/cmake/microxrcedds_agentTargets-release.cmake
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/cmake/config/microxrcedds_agentConfig.cmake
/home/intel/tb3_sbc_settings/Micro-XRCE-DDS-Agent/build/cmake/config/microxrcedds_agentConfigVersion.cmake
/usr/local/lib/libmicroxrcedds_agent.so.1
/usr/local/lib/libmicroxrcedds_client.a
/usr/local/lib/libmicroxrcedds_agent.so
/usr/local/lib/libmicroxrcedds_client.so.1.1.0
/usr/local/lib/libmicroxrcedds_client.so
/usr/local/lib/libmicroxrcedds_agent.so.1.1.0
/usr/local/share/microxrcedds_agent
/usr/local/share/microxrcedds_agent/cmake/microxrcedds_agentConfig.cmake
/usr/local/share/microxrcedds_agent/cmake/microxrcedds_agentTargets.cmake
/usr/local/share/microxrcedds_agent/cmake/microxrcedds_agentTargets-release.cmake
/usr/local/share/microxrcedds_client
/usr/local/share/microxrcedds_client/cmake/microxrcedds_clientConfigVersion.cmake
/usr/local/share/microxrcedds_client/cmake/microxrcedds_clientTargets-release.cmake
/usr/local/share/microxrcedds_client/cmake/microxrcedds_clientTargets.cmake
/usr/local/share/microxrcedds_client/cmake/microxrcedds_clientConfig.cmake
find: ‘/run/user/1000/gvfs’: Permission denied

OpusK, 2019-08-07:
@rafaelhuang-intel 

> The command of your provide not search any one, so I changed command to "sudo find / -name "microxrcedds"",

Oh sorry. I missed the option.

I think you can delete the files below.
```
/usr/local/lib/libmicroxrcedds_agent.so.1
/usr/local/lib/libmicroxrcedds_client.a
/usr/local/lib/libmicroxrcedds_agent.so
/usr/local/lib/libmicroxrcedds_client.so.1.1.0
/usr/local/lib/libmicroxrcedds_client.so
/usr/local/lib/libmicroxrcedds_agent.so.1.1.0
/usr/local/share/microxrcedds_agent
/usr/local/share/microxrcedds_client
```

rafaelhuang-intel, 2019-08-07:
@OpusK 

after delete all library files and re-build again, it's workable.

I tried (http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#setup)15.2 + 15.3, it's ok, but tried 15.2+15.4, cartographer cannot build map and popup error message "[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty", could you meet same problem?

intel@intel-desktop:~/tb3_sbc_settings$ ros2 launch turtlebot3_cartographer cartographer.launch.py
[INFO] [launch]: All log files can be found below /home/intel/.ros/log/2019-08-07-10-55-21-703317-intel-desktop-17899
[INFO] [launch]: Default logging verbosity is set to INFO

[INFO] [cartographer_node-1]: process started with pid [17909]
[INFO] [occupancy_grid_node-2]: process started with pid [17910]
[INFO] [rviz2-3]: process started with pid [17911]
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/turtlebot3_cartographer/share/turtlebot3_cartographer/config/turtlebot3_lds_2d.lua' for 'turtlebot3_lds_2d.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] I0807 10:55:22.000000 17909 configuration_file_resolver.cc:41] Found '/home/intel/turtlebot3_ws/install/cartographer/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[cartographer_node-1] I0807 10:55:22.000000 17909 submap_2d.cc:187] Added submap 1
[cartographer_node-1] I0807 10:55:22.000000 17909 map_builder_bridge.cc:132] Added trajectory with ID '0'.
[cartographer_node-1] F0807 10:55:22.519484 17909 msg_conversion.cc:126] Check failed: msg.intensities.size() == msg.ranges.size() (1 vs. 360) 
[cartographer_node-1] F0807 10:55:22.000000 17909 msg_conversion.cc:126] Check failed: msg.intensities.size() == msg.ranges.size() (1 vs. 360) 
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[cartographer_node-1] *** Check failure stack trace: ***
[cartographer_node-1]     @     0x7f3d7e5a40cd  google::LogMessage::Fail()
[cartographer_node-1]     @     0x7f3d7e5a5f33  google::LogMessage::SendToLog()
[cartographer_node-1]     @     0x7f3d7e5a3c28  google::LogMessage::Flush()
[cartographer_node-1]     @     0x7f3d7e5a6999  google::LogMessageFatal::~LogMessageFatal()
[cartographer_node-1]     @     0x5603b624a011  (unknown)
[cartographer_node-1]     @     0x5603b624a12d  (unknown)
[cartographer_node-1]     @     0x5603b623b2aa  (unknown)
[cartographer_node-1]     @     0x5603b61f775c  (unknown)
[cartographer_node-1]     @     0x5603b61f8721  (unknown)
[cartographer_node-1]     @     0x5603b622d9f4  (unknown)
[cartographer_node-1]     @     0x5603b622dc11  (unknown)
[cartographer_node-1]     @     0x7f3d7f954adf  rclcpp::executor::Executor::execute_subscription()
[cartographer_node-1]     @     0x7f3d7f955b25  rclcpp::executor::Executor::execute_any_executable()
[cartographer_node-1]     @     0x7f3d7f95bf2f  rclcpp::executors::SingleThreadedExecutor::spin()
[cartographer_node-1]     @     0x7f3d7f958c32  rclcpp::spin()
[... cut ...]
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty

OpusK, 2019-08-07:
@rafaelhuang-intel 

The issue with packages seems to take time, as mentioned in #460.
Thank you for your patience.

mkhansen-intel, 2019-08-12:
@OpusK @rafaelhuang-intel - this error being thrown by cartographer is causing it to crash:
```
msg.intensities.size() == msg.ranges.size() (1 vs. 360)
```
We've seen this also and it is a bug in the turtlebot3_lidar publisher. It isn't declaring the topic.intensities_size for the intensities data to be = 360. Changing that fixed the issue for us.

Since it's inside the tb3_sbc_settings.tar.bz2 tar file, I can't submit a PR with the fix, but the line of code I changed was in turtlebot3_lidar/main.cpp:
```
topic.intensities_size = 360; 
```
That fixes the problem for us.

rafaelhuang-intel, 2019-08-20:
@OpusK 

I've build a new pure environment and follow (http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)  to setup TB3, but I meet a problem of TB3 launch.

when I run script "./run.sh", I meet core dump (see below log) 

intel@intel-desktop:~/tb3_sbc_settings$ ./run.sh 

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: nohup: appending output to 'nohup.out'ignoring input and appending output to 'nohup.out'

nohup: ignoring input and appending output to 'nohup.out'
**./run.sh: line 13:  3425 Aborted                 (core dumped) nohup ./turtlebot3_lidar_xrce_client**

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

 use 'ps | grep "MicroXRCEAgent"' command for checking process ID
 kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
==========================

intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
 3416 pts/0    00:00:11 MicroXRCEAgent
 3417 pts/0    00:00:11 MicroXRCEAgent

**and than, I tried split launch command to debug (see below log)**

T1
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 1000000 -r tb3_fastrtps_profile.refs -v 5
Enter 'q' for exit

T2
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent udp -p 2019 -r tb3_fastrtps_profile.refs -v 5
Enter 'q' for exit
[1566272916.394301] info     | UDPServerLinux.cpp | init                     | running...             | port: 2019
[1566272916.394709] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 5
[1566272940.989220] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24
[1566272940.989287] info     | Root.cpp           | create_client            | create                 | client_key: 0xCCCCDDDD, session_id: 0x81
[1566272940.989316] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xCCCCDDDD, address: 127.0.0.1:130
[1566272940.989441] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 19
[1566272940.989550] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 120
[1566272940.991886] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xCCCCDDDD, participant_id: 0x001(1)
[1566272940.991970] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14
[1566272940.991998] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xCCCCDDDD, topic_id: 0x001(2), participant_id: 0x001(1)
[1566272940.992018] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xCCCCDDDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1566272940.992022] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13
[1566272940.992060] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14
[1566272940.992076] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14
[1566272940.992108] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13
[1566272940.992120] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13
[1566272940.992181] debug    | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xCCCCDDDD, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1566272940.992201] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14

T3
intel@intel-desktop:~/tb3_sbc_settings$ ./turtlebot3_lidar_xrce_client
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  open: Permission denied
Aborted (core dumped)

meet core dump in T3 to launch tb3 lidar, so I tried use root permission to launch, it's workable.
intel@intel-desktop:~/tb3_sbc_settings$ sudo ./turtlebot3_lidar_xrce_client
[sudo] password for intel: 
[... 39 lines cut ...]
[rviz2-1] [ERROR] [rviz2]: "base_scan" passed to lookupTransform argument source_frame does not exist. 
[rviz2-1] [ERROR] [rviz2]: "base_scan" passed to lookupTransform argument source_frame does not exist. 
[rviz2-1] [ERROR] [rviz2]: "base_scan" passed to lookupTransform argument source_frame does not exist. 
[rviz2-1] [ERROR] [rviz2]: "base_scan" passed to lookupTransform argument source_frame does not exist. 
[rviz2-1] [ERROR] [rviz2]: "base_scan" passed to lookupTransform argument source_frame does not exist. 
[... cut ...]
[rviz2-1] [ERROR] [rviz2]: Could not find a connection between 'odom' and 'base_scan' because they are not part of the same tree.Tf has two or more unconnected trees.
[rviz2-1] [ERROR] [rviz2]: Could not find a connection between 'odom' and 'base_scan' because they are not part of the same tree.Tf has two or more unconnected trees.
[rviz2-1] [ERROR] [rviz2]: Could not find a connection between 'odom' and 'base_scan' because they are not part of the same tree.Tf has two or more unconnected trees.

routiful, 2019-08-20:
@rafaelhuang-intel 
`base_scan` tf issue was occured from tf message filter.
You can check following issues [(1)](https://github.com/ros2/rviz/issues/359), [(2)](https://github.com/ros2/rviz/issues/332), [(3)](https://github.com/ros2/demos/issues/295)
It might be solved near future.

rafaelhuang-intel, 2019-08-20:
@routiful 

base_scan only show in Terminal 5. in my side, the first priority is TB3 launch problem,

intel@intel-desktop:~/tb3_sbc_settings$ ./run.sh

########################################################################
Run two Agent and one Client for LDS & OpenCR
########################################################################
nohup: nohup: appending output to 'nohup.out'ignoring input and appending output to 'nohup.out'

nohup: ignoring input and appending output to 'nohup.out'
./run.sh: line 13: 3425 Aborted (core dumped) nohup ./turtlebot3_lidar_xrce_client

Two MicroXRCEAgents and one MicroXRCEClient listed below are running in the background.
Please refer to the note below to check its execution status or to terminate it.

========== NOTE ==========
This script uses the nohup command to run multiple processes in the background.
Therefore, to terminate these processes, use the kill command like below

use 'ps | grep "MicroXRCEAgent"' command for checking process ID
kill [PID number]

Also, logs running in the background are stored in nohup.out in the current folder where this command was executed.
intel@intel-desktop:~/tb3_sbc_settings$ ps | grep "MicroXRCEAgent"
3416 pts/0 00:00:11 MicroXRCEAgent
3417 pts/0 00:00:11 MicroXRCEAgent

and than, I tried split launch command to debug (see below log)

T1
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 1000000 -r tb3_fastrtps_profile.refs -v 5
Enter 'q' for exit

T2
intel@intel-desktop:~/tb3_sbc_settings$ MicroXRCEAgent udp -p 2019 -r tb3_fastrtps_profile.refs -v 5
Enter 'q' for exit
[1566272916.394301] info | UDPServerLinux.cpp | init | running... | port: 2019
[1566272916.394709] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 5
[1566272940.989220] debug | UDPServerLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0x00000000, len: 24
[1566272940.989287] info | Root.cpp | create_client | create | client_key: 0xCCCCDDDD, session_id: 0x81
[1566272940.989316] info | UDPServerBase.cpp | on_create_client | session established | client_key: 0xCCCCDDDD, address: 127.0.0.1:130
[1566272940.989441] debug | UDPServerLinux.cpp | send_message | [** <> ] | client_key: 0xCCCCDDDD, len: 19
[1566272940.989550] debug | UDPServerLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0xCCCCDDDD, len: 120
[1566272940.991886] debug | ProxyClient.cpp | create_participant | participant created | client_key: 0xCCCCDDDD, participant_id: 0x001(1)
[1566272940.991970] debug | UDPServerLinux.cpp | send_message | [ <> ] | client_key: 0xCCCCDDDD, len: 14
[1566272940.991998] debug | ProxyClient.cpp | create_topic | topic created | client_key: 0xCCCCDDDD, topic_id: 0x001(2), participant_id: 0x001(1)
[1566272940.992018] debug | ProxyClient.cpp | create_publisher | publisher created | client_key: 0xCCCCDDDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1566272940.992022] debug | UDPServerLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0xCCCCDDDD, len: 13
[1566272940.992060] debug | UDPServerLinux.cpp | send_message | [ <> ] | client_key: 0xCCCCDDDD, len: 14
[1566272940.992076] debug | UDPServerLinux.cpp | send_message | [ <> ] | client_key: 0xCCCCDDDD, len: 14
[1566272940.992108] debug | UDPServerLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0xCCCCDDDD, len: 13
[1566272940.992120] debug | UDPServerLinux.cpp | recv_message | [==>> UDP <<==] | client_key: 0xCCCCDDDD, len: 13
[1566272940.992181] debug | ProxyClient.cpp | create_datawriter | datawriter created | client_key: 0xCCCCDDDD, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1566272940.992201] debug | UDPServerLinux.cpp | send_message | [ <> **] | client_key: 0xCCCCDDDD, len: 14

T3
intel@intel-desktop:~/tb3_sbc_settings$ ./turtlebot3_lidar_xrce_client
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injectorboost::system::system_error >'
what(): open: Permission denied
Aborted (core dumped)

meet core dump in T3 to launch tb3 lidar, so I tried use root permission to launch, it's workable.
intel@intel-desktop:~/tb3_sbc_settings$ sudo ./turtlebot3_lidar_xrce_client
[sudo] password for intel:
turtlebot3 lidar client setup complete!

and next , I launch teleop and tb3 rviz, and press "w/x/a/d" on keyboard, I can see the path will move and show in Rviz, but real TB3 no response.

routiful, 2019-08-21:
@rafaelhuang-intel 

TurtleBot3 ROS 2 Dashing version was released :)

Please refer to this [link](https://discourse.ros.org/t/tb3-ros-2-dashing-release/10364) and [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2)

Please attempt to upload new firmware. If you have any question or further issue, please reopen this or get a new ticket :)

closed 2019-08-21 by routiful

OpusK, 2019-08-22:
@rafaelhuang-intel 

As you know, TB3 no longer supports this method, so it seems good to close this issue.
May I close this issue?