ROBOTIS-GIT/turtlebot3 #463: when run SLAM simulate in gazebo, the map show in Rviz has drift while the TB3 is moving

rafaelhuang-intel, 2019-08-02:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Waffle

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] NUC6i7KYK

3. Which OS you installed in SBC?

    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] ROS2 Dashing

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] ROS2 Dashing

Follow http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2 to build TB3 for ROS2 dashing release, when run SLAM simulate in gazebo, the map show in Rviz has drift while the TB3 is moving

rafaelhuang-intel, 2019-08-02:
Attached capture and cartographer log:
log:
[TB3_SLAM_Simulate.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/3459729/TB3_SLAM_Simulate.txt)

capture:
https://drive.google.com/file/d/1gd1j7c12XoAxOZn1srL0_okGXPek6RF3/view?usp=sharing

rafaelhuang-intel, 2019-08-02:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/460

rafaelhuang-intel, 2019-08-22:
@routiful
 I tried on Dashing release, this issue still exist.

routiful, 2019-08-22:
@rafaelhuang-intel 
I was faced exactly same situation in gazebo. 
To adjust param related coefficient of friction is one of solution.
I will try to find more reliable param as soon as possible.

spragunr, 2020-10-27:
This is an old issue, but it seems to still be a problem in dashing. 

I was able to get mapping working in gazebo by changing this setting to `false` in `turtlebot3_lds_2d.lua`:

`TRAJECTORY_BUILDER_2D.use_imu_data = true`

I'm guessing the lack of simulate imu causes cartographer to go haywire.

AdoHaha, 2020-11-12:
had the same problem, @spragunr solution worked, but it really seems like a bug

rafaelhuang-intel, 2020-12-08:
had the same problem on Foxy release, @spragunr solution worked.

ROBOTIS-Will, 2021-01-29:
This issue has been fixed in the recent Foxy update.
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/135
Please let us know if the problem still exists.
Thank you.

closed 2021-03-10 by ROBOTIS-Will