ROBOTIS-GIT/turtlebot3 #464: cannot compile

r06942072, 2019-08-02:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
    - [x] Only do gazebo simulation

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order
I have tried both master branch and melodic-devel branch, and the result are same  
    -```git clone https://github.com/ROBOTIS-GIT/turtlebot3.git --branch melodic-devel```  

    - ```cd ~/catkin_ws```  
    - ```catkin_make```
 
7. Copy and Paste your error message on terminal

    - 
[image]

  
8. Please, describe detailedly what difficulty you are in 

    - I follow this instruction actually. 
https://programmer.help/blogs/ubuntu-18.04-lts-melodic-ros-configuration-turtlebot-3-running-gazebo-simulation.html

rjshim, 2019-08-05:
@r06942072 

Please refer to our official [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#install-dependent-ros-packages)

You don't have the turtlebot3_msgs package so you have to download by doing the following.
```sh
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git -b melodic-devel
```

Regards,
Ryan

rjshim, 2019-09-26:
This issue will be closed since there has been no response for a while. You can reopen this issue if you have any relevant questions. Thanks.

Ryan

closed 2019-09-26 by rjshim