ROBOTIS-GIT/turtlebot3 #465: DAE/STL files are too high resolution for reasonable viewing in RVIZ

SteveMacenski, 2019-08-05:
The Waffle plates are 16MB, each tire is a MB, astra is 8MB, these are way too high resolution for visualization in a reasonable load time in RVIZ, it slows down loading alot when enabling and will freeze the screen when added after launching.

Also documented here: https://github.com/ros-planning/navigation2/issues/1000

Can you please swap out the STL/DAEs in ROS2 with fewer polygons? There's not a huge need for that much detail for visualization

SteveMacenski, 2020-04-27:
Following up on this ticket too since it's marked "accepted"

ROBOTIS-Will, 2020-05-11:
We'd like to keep some of the distinctive shape for the TurtleBot3 platform. Still opened up for chances for more compact size, but definitely wheels and LDS should be shrunken down more.
I'll use these files for Foxy for now.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/bb0258f7f0cb58d36f8bce8bc590b3ebd147cebc/turtlebot3_description/meshes

SteveMacenski, 2020-05-11:
The plate meshes and electronics board meshes are also excessive. Downsampling that a bit wouldn’t lose shape, and its not like anyone is zooming into the board in gazebo to see how many USB ports it had 😉

ROBOTIS-Will, 2020-05-15:
Agreed, I was going to delete all the electronic boards on the platform. Let me double check the files and see if there's any unnecessary residues. Thanks!

SteveMacenski, 2020-05-15:
Awesome, let me know when you're done, I'm curious to know the size you end up with. If its reasonable enough, we can start using those rather than having copies in navigation2. Keep in mind this is just for visualization, not for 3D printing.

ROBOTIS-Will, 2020-12-09:
This issue will be closed as mesh files have been simplified and downsized for simulation.

closed 2020-12-09 by ROBOTIS-Will