ROBOTIS-GIT/turtlebot3 #466: turtlebot3 with different motors

CesMak, 2019-08-06:
Hey there,

I build an own robot (looks very similar to the Burger)

Hardware:
2x Dynamixel XH-540-W150 R
OpenCR Board
Connected to my Laptop

I did the manual of the turtlebot3 until [6.3.2Basic Operation](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-firmware-upload-for-tb3). 

Now when I press PUSH SW1 or PUSH SW2 no motor is turning.

My questions:
1. On what operating Mode (11) do I have to configure the motors? [PWM, Velocity, Position?]
2. I assigned the motors to have ID(1,2) and also changed the code here:

```
#define LEFT                             2 // ID before 0
#define RIGHT                            1
```

3. My Baudrate of the Motors was set via R+ to 3M and I use protocol version 2.0 do I have to set this somewhere inside the turtlebot_core code? I could not find it.
4. Do I have to connect the battery or should the wheels also turn if I connect directly to power power supply?
5. I also tried to connect via USB directly to the Laptop and then did:
roslaunch turtlebot3_bringup turtlebot3_robot.launch
-> (I commented the Lidar node to be not launched) 
-> I still got this error:
```
... logging to /home/markus/.ros/log/3bef905c-b87b-11e9-9023-48d2244c225b/roslaunch-markus-11915.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:38695/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 57600
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)

auto-starting new master
process[master]: started with pid [11925]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3bef905c-b87b-11e9-9023-48d2244c225b
process[rosout-1]: started with pid [11938]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [11947]
process[turtlebot3_diagnostics-3]: started with pid [11956]
[INFO] [1565117505.152001]: ROS Serial Python Node
[INFO] [1565117505.160838]: Connecting to /dev/ttyACM0 at 57600 baud
[ERROR] [1565117522.271209]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
```
-> I adjusted my baudrate as follows:
```
  DEBUG_SERIAL.begin(57600);

  // Initialize ROS node handle, advertise and subscribe the topics
  nh.initNode();
  nh.getHardware()->setBaud(57600);  // before:115200
```
-> I also checked the port it is ACM0 (I know this because if I flash the code with arduino I see this port and if I flash it with this port it works)

Would be happy if you could support me.

rjshim, 2019-08-07:
@CesMak 

1) Velocity
2) Do not change the source code but set your motor IDs to 1(Left), 2(Right) 
3) you can change the baud rate in the following file 
~/(path_to_turtlebot3)/include/turtlbot3/turtlebot3_motor_driver.h 
But it won't make any difference so just set your motor baud rates to 1000000. 
4) Ok for both. Read how to setup section in our eManual. 
5) https://github.com/ROBOTIS-GIT/turtlebot3/issues/130#issue-297909949
If you don't fix your problem with the answer from the link above, you will need to study [link](http://wiki.ros.org/rosserial) 

We do not support each person's own customized robot. If you would like to learn more, please build a robot with the same setup as our TB3, follow the uploaded tutorials on our eManual and ask us if you have any questions.

Regards,
Ryan

CesMak, 2019-08-07:
Thanks its now working :) 
You can close it.

`
markus@markus:~/ros_space/turtlebot3_code$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/markus/.ros/log/4497c566-b93a-11e9-a737-48d2244c225b/roslaunch-markus-6845.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:37073/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 57600
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)

auto-starting new master
process[master]: started with pid [6865]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4497c566-b93a-11e9-a737-48d2244c225b
process[rosout-1]: started with pid [6878]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [6881]
process[turtlebot3_diagnostics-3]: started with pid [6882]
[INFO] [1565199554.504908]: ROS Serial Python Node
[INFO] [1565199554.512650]: Connecting to /dev/ttyACM0 at 57600 baud
[INFO] [1565199556.630751]: Note: publish buffer size is 512 bytes
[INFO] [1565199556.631269]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1565199556.634583]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1565199556.719653]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1565199556.722936]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1565199556.760309]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1565199556.763580]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1565199556.767570]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1565199556.771000]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1565199557.767167]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1565199557.773291]: Note: subscribe buffer size is 512 bytes
[INFO] [1565199557.773872]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1565199557.779130]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1565199557.783598]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1565199557.788537]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1565199559.742790]: Setup TF on Odometry [odom]
[INFO] [1565199559.745067]: Setup TF on IMU [imu_link]
[INFO] [1565199559.747225]: Setup TF on MagneticField [mag_link]
[INFO] [1565199559.749091]: Setup TF on JointState [base_link]
[INFO] [1565199559.755820]: --------------------------
[INFO] [1565199559.757995]: Connected to OpenCR board!
[INFO] [1565199559.759717]: This core(v1.2.3) is compatible with TB3 Burger
[INFO] [1565199559.761462]: --------------------------
[INFO] [1565199559.763057]: Start Calibration of Gyro
[INFO] [1565199562.303969]: Calibration End
[ERROR] [1565199562.313248]: Message from device dropped: message larger than buffer.
[ERROR] [1565199562.342436]: Message from device dropped: message larger than buffer.

`

CesMak, 2019-08-07:
Hm, I now tried to do 

`
 roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch 
`

on remote pc. If I send any commands no motor is turning....
However if I echo on /imu etc. I get messages on remote and I also see
the cmd_vel commands if I send them on remote pc and echo them on turtlebotpc.

The only error I get on tb3_pc:

`Message from device dropped: message larger than buffer`

Now the strange thing is, that if I press the SW1 or SW2 button without rosserial communication to turtlebot3 pc on, the motors are turning and it works.

What do I miss here? Any help would be great.

rjshim, 2019-08-08:
@CesMak 

[This](https://github.com/ROBOTIS-GIT/turtlebot3/issues/160#issue-307090677) might help you. 

Regards,
Ryan

CesMak, 2019-08-10:
Thanks that worked!

rjshim, 2019-08-11:
@CesMak 

Glad it somehow helped you. I will close this issue.

Ryan

closed 2019-08-11 by rjshim