ROBOTIS-GIT/turtlebot3 #467: Release for melodic

OTL, 2019-08-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?
    - Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    -  Raspberry Pi 3

3. Which OS you installed in SBC?

    - Ubuntu 18.04.2

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    `roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml`
 
7. Copy and Paste your error message on terminal

Error1:

```
Warning: Invalid argument "/map" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like:         
         at line 134 in /home/ogura/catkin_ws/src/geometry2/tf2/src/buffer_core.cpp                                                  
```

Error2:

```
  Node: /move_base
Time: 11:23:10.702144405 (2019-08-07)
Severity: Warn
Published Topics: /cmd_vel, /move_base/current_goal, /move_base/goal, /rosout

Timed out waiting for transform from /base_footprint to /map to become available before running costmap, tf error: . canTransform returned after 10.0086 timeout was 10.

Location:
/home/ogura/catkin_ws/src/navigation/costmap_2d/src/costmap_2d_ros.cpp:Costmap2DROS::Costmap2DROS:101

----------------------------------------------------------------------------------------------------

```

8. Please, describe detailedly what difficulty you are in 

I'm using turtlebot3 with melodic/ubuntu18.04, and the navigation failed with above error message and there are no costmap. That is caused by tf2 update, and has been solved by 69b4563d754ad already.
But i struggled with the problem for several hours because I used melodic binary and 'melodic-devel' branch of the source of this repo.
The turtlebot3_navigation on melodic is broken because of the tf2 update, so I recommend to release the fix for melodic if possible.

ROBOTIS-Kayman, 2019-08-19:
Thank you for the notice.
We'll take care of it soon.

Regards,
Kayman

robotpilot, 2019-08-20:
Hi @OTL 
Sorry for late. We are going to work on bloom-releasing today.
For ROS 1 Melodic Morenia:
-> https://github.com/ros/rosdistro/pull/22059
-> https://github.com/ROBOTIS-GIT/turtlebot3/releases/tag/1.2.2

closed 2019-09-08 by robotpilot