ROBOTIS-GIT/turtlebot3 #469: Latest dashing robot.launch.py doesn't create scan or scan_half topics

oneattosecond, 2019-08-07:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ X ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [X] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: ???
    - Firmware version: 1.0.2
 
6. Write down the commands you used in order

    - ros2 launch turtlebot3_bringup robot.launch.py
 
7. Copy and Paste your error message on terminal

    - No error on terminal
  
8. Please, describe detailedly what difficulty you are in 

    - with latest sources, robot.launch.py doesn't result in scan or scan_half topics
    - this means cartographer, SLAM, navigation do not work at all
    - I went back and built turtlebot3.repos from ~1 month ago (before switch to dashing branches for nav2, etc) and the robot.launch.py from 1 month ago works - scan topic is published... IMU data flows, cartographer works

oneattosecond, 2019-08-13:
This appears to be intentionally removed by this commit:

https://github.com/ROBOTIS-GIT/turtlebot3/commit/9e02d2665cf88fe85d54dc646e766c77d811efda

I'm trying to understand why? scan and scan_half topics are critical to the functionality of cartographer and SLAM, aren't they?

routiful, 2019-08-23:
This issue was solved after released new firmware. If you have any questions or further issue, please get a new ticket.

closed 2019-08-23 by routiful