ROBOTIS-GIT/turtlebot3 #470: Plans to support Ubuntu 18.04 and Melodic?

pitosalas, 2019-08-09:
Our teaching semester is about to start and I am configuring our robots and laptops for the incoming class. I would like to use Ubuntu 18.04 and Melodic. Can you let us know, when do you expect to update http://emanual.robotis.com/docs/en/platform/turtlebot3/overview/ instructions to use Ubuntu 18.04 and Melodic?

In particular when will you release a supported Melodic version of http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/

Thanks!

ROBOTIS-Kayman, 2019-08-19:
TB3 packages are ready to support Melodic version, but it has not announce officially to support Melodic due to support issues with associated other packages.
For this reason, there are no emanual update plans yet.

**6.1.2**
In case of PC-Setup for melodic, the installation script is ready at [this link](https://github.com/ROBOTIS-GIT/robotis_tools/blob/master/install_ros_melodic.sh).
```
$ sudo apt-get update
$ sudo apt-get upgrade
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_melodic.sh && chmod 755 ./install_ros_kinetic.sh && bash ./install_ros_melodic.sh
```

**6.1.3**
Install Dependent ROS Packages(gmapping does not support melodic.)
```
sudo apt-get install ros-melodic-joy \
ros-melodic-teleop-twist-joy \
ros-melodic-teleop-twist-keyboard \
ros-melodic-laser-proc \
ros-melodic-rgbd-launch \
ros-melodic-depthimage-to-laserscan \
ros-melodic-rosserial-arduino \
ros-melodic-rosserial-python \
ros-melodic-rosserial-server \
ros-melodic-rosserial-client \
ros-melodic-rosserial-msgs \
ros-melodic-amcl \
ros-melodic-map-server \
ros-melodic-move-base \
ros-melodic-urdf \
ros-melodic-xacro \
ros-melodic-compressed-image-transport \
ros-melodic-rqt-image-view \
ros-melodic-navigation \
ros-melodic-interactive-markers
```

The rest of the PC-Setup is the same as the emanual.

robotpilot, 2019-08-20:
FYI, currently, 23 TurtleBot3 packages are available as binary packages for ROS 1 Melodic.
http://repositories.ros.org/status_page/ros_melodic_default.html?q=turtlebot3

closed 2019-09-08 by robotpilot