ROBOTIS-GIT/turtlebot3 #478: TB3 map creating

Adel-kn, 2019-08-20:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello everyone,
my question is if one can create the map using SLAM in more than one time, that means accomplishing an already created map to extend it.
Thanks in advance

rjshim, 2019-08-21:
@Adel-kn 

Please ask your question at the following [link](https://wiki.ros.org/gmapping) as it is not a tb3 related question but more like a question about the SLAM package.

Regards,
Ryan

rjshim, 2019-08-21:
@Adel-kn

Please ask your question at the following link as it is not a tb3 related question but more like a question about the SLAM package.

Regards,
Ryan

Adel-kn, 2019-08-21:
@rjshim 
thanks Ryan

closed 2019-08-22 by rjshim