ROBOTIS-GIT/turtlebot3 #479: latest dashing TB3 robot.launch.py throws RCLError due to invalid node context

oneattosecond, 2019-08-21:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [X] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: latest posted to wiki today
    - Firmware version: latest posted to wiki today

7. Write down the commands you used in order

    - After updating Raspberry Pi using latest wiki instructions, and building latest TB3 code released ~today -- robot.launch.py crashes 100% on launch

8. Copy and Paste your error message on terminal

[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init TurtleBot3 Node Main
[ERROR] [hlds_laser_publisher-2]: process has died [pid 15980, exit code 255, cmd '/home/user/tb3_dashing_src/install/hls_lfcd_lds_driver/lib/hls_lfcd_lds_driver/hlds_laser_publisher __node:=hlds_laser_publisher __params:=/tmp/launch_params_xcgbp31y'].
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [PortHandlerLinux::SetupPort] Error opening serial port!
[turtlebot3_ros-3] [ERROR] [DynamixelSDKWrapper]: Failed to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [ERROR] [DynamixelSDKWrapper]: Failed to initialize SDK handlers
[turtlebot3_ros-3] [ERROR] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /home/user/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:476
[ERROR] [turtlebot3_ros-3]: process has died [pid 15981, exit code -6, cmd '/home/user/tb3_dashing_src/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/user/tb3_dashing_src/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle_pi.yaml'].

9. Please, describe detailedly what difficulty you are in

    - My TB3 doesn't work at all on latest sources

routiful, 2019-08-22:
@oneattosecond 

> [turtlebot3_ros-3] [PortHandlerLinux::SetupPort] Error opening serial port!

Port was not openned. Try to apply rules file by using below commands

```
$ cd ~/turtlebot3_ws/src/turtlebot3/turtlebot3_bringup
$ sudo cp ./99-turtlebot3-cdc.rules /etc/udev/rules.d/
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
```

oneattosecond, 2019-08-22:
@routiful thanks for the suggestion, this was not listed on wiki last night. Unfortunately, I have followed the instruction and still get the same exact error.

Can you kindly upload your whole Ubuntu image to solve any inconsistency in setup?

Please note the wiki has a typo, it says to "cd ~/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_bringup"

But the correct path (at least on my device) is ~/turtlebot_ws/src/turtlebot3/turtlebot3_bringup/

oneattosecond, 2019-08-22:
Here is the backtrace in GDB, in case it proves useful... I tried rebooting SBC few times, checked every step on the wiki, and still hit this error.

(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff5027801 in __GI_abort () at abort.c:79
#2  0x00007ffff5a1a957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5a20ab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5a20af1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5a20aaa in std::rethrow_exception(std::__exception_ptr::exception_ptr) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff6d14e55 in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_t const*, void (*)()) ()
   from /home/bro/ros2_ws/install/rclcpp/lib/librclcpp.so
#7  0x00007ffff6df4b96 in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_t const&) () from /home/bro/ros2_ws/install/rclcpp/lib/librclcpp.so
#8  0x000055555567c3f0 in rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t const&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > > const&) ()
#9  0x000055555567c2b5 in void __gnu_cxx::new_allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > >::construct<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&>(rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >*, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&) ()
#10 0x000055555567c007 in void std::allocator_traits<std::allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > > >::construct<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&>(std::allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > >&, rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >*, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::sh---Type <return> to continue, or q <return> to quit---
ared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&) ()
#11 0x000055555567bc65 in std::_Sp_counted_ptr_inplace<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >, std::allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&>(std::allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&) ()
#12 0x000055555567b613 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> >, std::allocator<rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t&, rclcpp::PublisherEventCallbacks const&, std::shared_ptr<std::allocator<sensor_msgs::msg::BatteryState_<std::allocator<void> > > >&>(std::_Sp_make_shared_tag, rclcpp::Publisher<sensor_msgs::msg::BatteryState_<std::allocator<void>  [... cut ...] s<char>, std::allocator<char> >&>(std::allocator<robotis::turtlebot3::TurtleBot3> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
#33 0x00005555555dab55 in std::shared_ptr<robotis::turtlebot3::TurtleBot3> std::make_shared<robotis::turtlebot3::TurtleBot3, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
#34 0x00005555555da437 in main ()

routiful, 2019-08-23:
@oneattosecond 

> Can you kindly upload your whole Ubuntu image to solve any inconsistency in setup?

We can't upload Ubuntu image to be used commercial purpose. 

> Please note the wiki has a typo, it says to "cd ~/turtlebot3_ws/src/turtlebot3/turtlebot3/turtlebot3_bringup"

Thank your for your point. I have been modified emanual.

Have you ever launched `robot.launch.py` on your remote PC?
I found that the bringup section in [emanual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#bringup-turtlebot3) was wrong.

turtlebot3_bringup has to be launched on Raspberry Pi 3.

Sorry for inconvenience. If you have further questions, please feel free to new comments or get new ticket.

oneattosecond, 2019-08-23:
@routiful robot.launch.py is always supposed to be launched on Raspberry Pi? Or is this new to the dashing package support? I've been running robot.launch.py on remote PC for long time, is this incorrect?

routiful, 2019-08-23:
@oneattosecond 
No, `robot.launch.py` has to be launched on Raspberry Pi. 
Even [ROS 1](http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup-a-turtlebot3), the launch file was launched on Raspberry Pi.
I think you used to `ssh` command to communicate between remote pc and Raspberry Pi

oneattosecond, 2019-08-23:
OK, thank you for fixing the wiki to make this clear. I have only used TB3 with ROS2 so this was unclear. Will try today and report if I am up and running...

oneattosecond, 2019-08-23:
no longer see the crash when running robot.launch.py on Raspberry Pi, per the updated instructions.

I am able to teleop but cartographer and navigation do not work. Closing this issue and filing others soon.

closed 2019-08-23 by oneattosecond

routiful, 2019-08-26:
@oneattosecond Can you share error messages in new ticket when you launch cartographer and navigation packages? Then I will help you to solve that issues.