ROBOTIS-GIT/turtlebot3 #480: tb3_ros2 package failed to compile

EndlessLoops, 2019-08-27:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [+] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [+] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [+] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [+ Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

   `ubuntu@ubuntu:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> cartographer_ros_msgs
Starting >>> dynamixel_sdk
Finished <<< dynamixel_sdk [0.71s]                                                                             
Starting >>> turtlebot3_msgs
Finished <<< turtlebot3_msgs [3.00s]                                                                               
Starting >>> hls_lfcd_lds_driver
Finished <<< hls_lfcd_lds_driver [0.49s]                                                                                
Starting >>> turtlebot3_description
Finished <<< turtlebot3_description [0.41s]                                                                           
Starting >>> turtlebot3_node
Finished <<< cartographer_ros_msgs [4.71s]
Starting >>> cartographer_ros
Finished <<< turtlebot3_node [0.72s]                                                                    
Starting >>> turtlebot3_navigation2
--- stderr: cartographer_ros                                                                            
CMake Error at CMakeLists.txt:30 (find_package):
  By not providing "Findcartographer.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cartographer", but CMake did not find one.

  Could not find a package configuration file provided by "cartographer" with
  any of the following names:

    cartographerConfig.cmake
    cartographer-config.cmake

  Add the installation prefix of "cartographer" to CMAKE_PREFIX_PATH or set
  "cartographer_DIR" to a directory containing one of the above files.  If
  "cartographer" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< cartographer_ros	[ Exited with code 1 ]
--- stderr: turtlebot3_navigation2                                     
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findnavigation2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "navigation2", but CMake did not find one.

  Could not find a package configuration file provided by "navigation2" with
  any of the following names:

    navigation2Config.cmake
    navigation2-config.cmake

  Add the installation prefix of "navigation2" to CMAKE_PREFIX_PATH or set
  "navigation2_DIR" to a directory containing one of the above files.  If
  "navigation2" provides a separate development package or SDK, be sure it
  has been installed.

---
Aborted  <<< turtlebot3_navigation2

Summary: 6 packages finished [6.72s]
  1 package failed: cartographer_ros
  1 package aborted: turtlebot3_navigation2
  2 packages had stderr output: cartographer_ros turtlebot3_navigation2
  6 packages not processed
`

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in
I have successfully installed ros2 dashing using source code and I installed the tb3_ros2 package on the PC according to the [tutorial](https://github.com/ROBOTIS-GIT/emanual/blob/master/docs/en/platform/turtlebot3/ros2.md).

I have installed navigation2 and cartographer according to the tutorial, but it still fails to compile.

```
# Install Cartographer
$ sudo apt install ros-dashing-cartographer
# Install Navigation2
$ sudo apt install ros-dashing-navigation2
```

routiful, 2019-08-28:
@EndlessLoops 
Did you try to use `apt update` command?

EndlessLoops, 2019-08-30:
@routiful 
I tried but it could‘t work . 
I try install navigation2 and cartographer by source code now.

EndlessLoops, 2019-08-30:
`source /opt/ros/dashing/setup.bash`

routiful, 2019-08-30:
@EndlessLoops Did you solved?

EndlessLoops, 2019-08-30:
I got the new problem.
```
--- stderr: cartographer_ros                               
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
In file included from /home/ubuntu/turtlebot3_ws/src/utils/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc:17:0:
/home/ubuntu/turtlebot3_ws/src/utils/cartographer_ros/cartographer_ros/./cartographer_ros/map_builder_bridge.h:25:10: fatal error: cartographer/common/mutex.h: No such file or directory
 #include "cartographer/common/mutex.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ubuntu/turtlebot3_ws/install/cartographer_ros_msgs/include/cartographer_ros_msgs/msg/landmark_entry.hpp:7:0,
                 from /home/ubuntu/turtlebot3_ws/src/utils/cartographer_ros/cartographer_ros/./cartographer_ros/msg_conversion.h:26,
                 from /home/ubuntu/turtlebot3_ws/src/utils/cartographer_ros/cartographer_ros/cartographer_ros/msg_conversion.cc:17:
/home/ubuntu/turtlebot3_ws/install/cartographer_ros_msgs/include/cartographer_ros_msgs/msg/landmark_entry__struct.hpp:29:10: fatal error: geometry_msgs/msg/pose__struct.hpp: No such file or directory
 #include "geometry_msgs/msg/pose__struct.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/cartographer_ros.dir/cartographer_ros/msg_conversion.cc.o] Error 1
make[1]: *** [CMakeFiles/cartographer_ros.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< cartographer_ros	[ Exited with code 2 ]

Summary: 11 packages finished [8.79s]
  1 package failed: cartographer_ros
  1 package had stderr output: cartographer_ros
  2 packages not processed
```

EndlessLoops, 2019-09-03:
After I reinstall ubuntu 18.04 ，everything works fine.

closed 2019-09-03 by EndlessLoops