ROBOTIS-GIT/turtlebot3 #483: Odometry changed

CesMak, 2019-09-02:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] AS ROCK 

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

Hey there my odometry somewho behaves really strange:

If I increase my angluar velocity using a and d in the teleop_node, the wheels turn both in the same direction and when echoing the odom topic it says I am turning but my wheels both turn in the same direction on the real robot. 

If I increase my linear velocity it is the other way round (as above)

**What might be the reason for this? 
Did I connect my motors not correctly to the OpenCR Board?**

In my config turtlebot3_core_config.h I did not change anything:
```

#define LEFT                             0 // ID before 0
#define RIGHT                            1

#define LINEAR                           0 // before 0
#define ANGULAR                          1 // before 1
```

rjshim, 2019-09-05:
@CesMak 

1. Test the same process with Gazebo and see if you have the same problem. 
2. Do the following [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/setup/#setup) again (especially OpenCR setup section) 

Let us know if it still doesn't work
Ryan

CesMak, 2019-09-05:
I tried these instructions:

Press and hold PUSH SW 1 for a few seconds to command the robot to move 30 centimeters (about 12 inches) forward.
Press and hold PUSH SW 2 for a few seconds to command the robot to rotate 180 degrees in place.

For me If I push SW2 (outer button) my robot moves 30 cm. 
For me If I push SW1 (inner button) my robot rotates 180 deg. 

so it is all the other way round. 

**There should be a quick fix for this or?**

rjshim, 2019-09-05:
@CesMak 

1.Test if your left and right wheel motors move the same way when you give position commands to them. Use Dynamixel Wizard2.0 when you test it. [link](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/).
2. Check if you assembled your robot wrong (especially wheel motor directions). [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/hardware_setup/#part-list)
3. There is no way I can see what modifications you made on your source code. I suggest you reset everything and reinstall the firmware and see if you still have the same problem. [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup)

Follow the instruction above and let me know the result, 
Ryan

CesMak, 2019-09-08:
Thanks I just reversed the correct motor and now everything works fine :) 

You can close this.

rjshim, 2019-09-09:
Good work !! :+1:

closed 2019-09-09 by rjshim