ROBOTIS-GIT/turtlebot3 #485: TB3_ROS2:Get error "terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'" innavigation2

EndlessLoops, 2019-09-04:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

-  Burger

2. Which ROS is working with TurtleBot3?

- ROS 2 Dashing Diademata

3. Which SBC(Single Board Computer) is working on TurtleBot3?

- Raspberry Pi 3

4. Which OS you installed in SBC?

- Ubuntu Server 18.04 or later

5. Which OS you installed in Remote PC?

- Ubuntu 18.04 LTS (Bionic Beaver)

6. Write down software version and firmware version

- Software version: [x.x.x]
- Firmware version: [x.x.x]

7. Write down the commands you used in order

- SBC

    `$ ros2 launch turtlebot3_bringup robot.launch.py`

- Remote PC

    `$ ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/map.yaml`

8. Copy and Paste your error message on terminal

- SBC

```
ubuntu@TurtleBot3_ROS2:~$ ros2 launch turtlebot3_bringup robot.launch.py

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2019-09-04-10-53-19-533472-TurtleBot3_ROS2-3565
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [3575]
[INFO] [hlds_laser_publisher-2]: process started with pid [3576]
[INFO] [turtlebot3_ros-3]: process started with pid [3577]
[robot_state_publisher-1] Initialize urdf model from file: /home/ubuntu/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] got segment base_footprint
[robot_state_publisher-1] got segment base_link
[robot_state_publisher-1] got segment base_scan
[robot_state_publisher-1] got segment caster_back_link
[robot_state_publisher-1] got segment imu_link
[robot_state_publisher-1] got segment wheel_left_link
[robot_state_publisher-1] got segment wheel_right_link
[robot_state_publisher-1] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-1] Adding fixed segment from base_link to caster_back_link
[robot_state_publisher-1] Adding fixed segment from base_link to imu_link
[robot_state_publisher-1] Adding fixed segment from base_link to base_scan
[robot_state_publisher-1] Adding moving segment from base_link to wheel_left_link
[... 85 lines cut ...]
[world_model-4] [INFO] [global_costmap.global_costmap]: Checking transform
[rviz2-9] [ERROR] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : active samplers with a different type refer to the same texture image unit
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00831 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00147 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00115 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00301 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.0048 timeout was 1.
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00529 timeout was 1.
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565617.190 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00627 timeout was 1.
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565617.390 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565617.590 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565617.791 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565617.991 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565618.192 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565618.392 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565618.593 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565618.793 for reason(0)
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 1567565618.995 for reason(0)
[world_model-4] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00391 timeout was 1.
```

9. Please, describe detailedly what difficulty you are in

 I have successfully built a map with cartographer in ROS2 but  I get the problem in navigation2.

glmnet, 2019-09-18:
Me too, followed manual steps as close as possible

rjshim, 2019-10-07:
[https://github.com/ROBOTIS-GIT/turtlebot3/issues/426](url)

rjshim, 2019-10-28:
@EndlessLoops  @glmnet 

Would you download all the latest packages (for opencr, tb3, remote pc) and reset all configurations and let me know the result? 

Regards,
Ryan

EndlessLoops, 2019-10-28:
@rjshim 
I will retest it in the next few days and tell you the result. Thanks for your reply.

closed 2020-04-26 by EndlessLoops