ROBOTIS-GIT/turtlebot3 #486: Magnetometer Sensitivity

tfe104270, 2019-09-04:
Hi,

In turtlebot3_sensor.h, the magnetometer sensitivity (micro tesla per LSB) is incorrect.
`#define MAG_FACTOR                        6e-7`

According to [ak8963 datasheet](https://www.akm.com/akm/en/file/datasheet/AK8963C.pdf), in 16 bit mode, the sensitivity is 
> 0.15 uT / LSB

The  `turtlebot3_sensor.h` applied 14 bit mode settings  and is incorrect. Please help to correct it, thanks!

rjshim, 2019-09-10:
@tfe104270 

Would you tell us more about your question? What does the comment below mean? What made you think applying 14-bit mode settings is incorrect. 
> The turtlebot3_sensor.h applied 14-bit mode settings and is incorrect

Regards,
Ryan

rjshim, 2019-09-26:
@tfe104270 

We discussed it in the team and changed the source code as your advice [Link](https://github.com/ROBOTIS-GIT/OpenCR/pull/204). Thank you for your contribution.
Regards,
Ryan

closed 2019-09-26 by rjshim