ROBOTIS-GIT/turtlebot3 #489: Gmapping, overlapped maps

Adel-kn, 2019-09-06:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello,
I am creating a map using gmapping (SLAM) but during the operation, I get  overlapped
maps and the robot lost its pose many times and get the message "Scan Matching Failed, using odometry. Likelihood=0 "
Did anyone meet this problem before and found the solution?
Thank in advance 

[image]

Adel-kn, 2019-09-06:
The Problem is SOLVED.
The problem was that I had not enough obstacles in the area where the map is corrupted. This area was wide enough for the robot to lose its POSE which leads to this problem. Then I only added some obstacles (carton boxes) in the empty area (the middle of the square in the map) and then I got an accurate map

ROBOTIS-Kayman, 2019-09-09:
Hi,
Thank you for sharing your experience.

Regards,
Kayman

closed 2019-09-09 by ROBOTIS-Kayman