ROBOTIS-GIT/turtlebot3 #490: Turtlebot3_Fake.rviz superpositioning under odom

Somusundram, 2019-09-07:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] etc (Dell Inspiron)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [x] etc (SIMULATION)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - See attached pic

8. Copy and Paste your error message on terminal

    - Error in picture format (rviz)
[image]

9. Please, describe detailedly what difficulty you are in

- I ran Turtlebot3_Fake.rviz .It seems 2 robot appears at same time but it's switching position. This happens when i execute Gazebo turlebot3_world.launch. Another thing is that if i execute the roslaunch turtlebot3_fake.launch, the robot (burger) appears to be empty and TF throws "no tf data. actual error: fixed frame [odom] does not exist" . Does this has to do with missing or corrupted installation?

rjshim, 2019-09-09:
@Somusundram 

1. 11.1 Fake Node and 11.2 TB3 Gazebo are separate different examples. Run only either of 11.1 or 11.2 and test it again. 

2. Would you attach a screenshot of your terminal with error code so that we can figure out what the problem is?
> roslaunch turtlebot3_fake.launch, the robot (burger) appears to be empty and TF throws "no tf data. actual error: fixed frame [odom] does not exist" . 

Regards,
Ryan

rjshim, 2019-09-26:
This issue will be closed since there has been no response for a while. You can reopen this issue if you have any relevant questions. Thanks.

Ryan

closed 2019-09-26 by rjshim