ROBOTIS-GIT/turtlebot3 #491: Navigation

Adel-kn, 2019-09-09:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello everyone,
Did anyone meet a problem by navigation that the costmap (local and global) does not match the static map and then get many errors and unexpected behavior during navigation?
 Thanks in adance
[image]

rjshim, 2019-09-10:
@Adel-kn 

After a rviz window pops up, you need to set the approximate initial pose (position and orientation) of the robot by using '2D Pose Estimate'. Then, when you set a goal pose by using '2D Nav goal', the robot will find its more accurate pose while navigating.

Try it and let us know if you still have the same problem, 
Ryan

Adel-kn, 2019-09-10:
Hello Ryan,
thanks for the answer.
I already have set the initial pose using " '2D Pose Estimate'", and navigated to a goal.
At the beginning after setting the initial pose everything was ok, but during the driving to the goal, I get this problem and should reinitialize the pose and then after a while during the navigation I get the problem again and again.
Any Idea?

Adel-kn, 2019-09-16:
@rjshim 
Do you have another idea to solve this problem ?

rjshim, 2019-09-16:
@Adel-kn 

I cannot think of any reasons for the problem, but I would suggest you 

Step 1. Follow our [tutorial](http://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#slam) and check if it works fine. If you have any problems, ask us! 
Step 2. Now you make your own map in the same way as Step 1, but you would want to check when you have different results from when you did Step 1. if you get different results, think about what you did differently and why that gave you different results. 

Honestly, there is no way we can really help if you just ask questions like I did my own thing and it doesn't work.

Regards,
Ryan

Adel-kn, 2019-09-17:
@rjshim 
thank you for your response
Actually, I tried to navigate with another map which I created but still have the same problem.
But when I created a map using a virtual environment (gazebo) and navigated with it then everything is fine.
So I am thinking that the problem may be related to a frequency, but I am not sure about that

rjshim, 2019-09-18:
@Adel-kn 

Would you try the [following link](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc)?

Regards,
Ryan

Adel-kn, 2019-09-18:
Helo @rjshim 
Thanks for the advice but It is not the problem

Adel-kn, 2019-09-18:
Here are the tf_tree and the node_grapgh, may they could help 
[image]
[image]

rjshim, 2019-09-19:
@routiful Could I get some advice on this issue? I cannot think of any possible reasons for the problem.

rjshim, 2019-09-19:
@Adel-kn @routiful 
I just tested the SLAM and navigation packages and they worked fine. 

[image]

rjshim, 2019-09-19:
@Adel-kn 

Would you try making a map of narrow corridors or small room and post it here again? Check it your robot slips on the ground surface while navigating. 

Regards,
Ryan

Adel-kn, 2019-09-20:
Hello @rjshim  
thanks for all your responses and help.
I think my problem was not related to the robot or the navigation stack themselves but it is related to the environment where I am executing the navigation. That means big environments or long corridors without distinguishable landmarks or with only a few of them will lead to a bad localization of the robot (the robot actually loses its pose within the environment).
And the solutions to this problem in my opinion are:
1- Once the robot loses its pose we should help it by giving it its actual pose using "pose estimation in Rviz".
2- trying the do the navigation properly or in a smart way by driving the robot to the wanted goal in such a way it has always some special landmarks in its way to the goal
3- If we want to build our environment for doing SLAM and navigation we should be aware to use many landmarks and they should not be the identic in the shape or specification that means to make a distinguishable environment (in the robot point of view) as possible as we can

rjshim, 2019-09-21:
@Adel-kn 

Good work! Glad you figured out the problem and thank you for sharing it with us.

Let us know if you have any other questions while using our platform and I will close this issue :) 
Ryan

closed 2019-09-21 by rjshim