ROBOTIS-GIT/turtlebot3 #495: Error in saving the map - Dashing Diademata

priyablue, 2019-10-01:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x ] Laptop

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [x] Ubuntu 18.04

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] working on simulation

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
    - ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
    - ros2 run turtlebot3_teleop teleop_keyboard
   once map generated, ros2 run nav2_map_server map_saver -f ~/map

8. Copy and Paste your error message on terminal

    - error occurs when i try to save the map generated.  
[WARN] [map_saver_cli]: Ignoring unrecognized argument '/home/bluetronics/map'
[INFO] [map_saver]: Waiting for the map
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...

9. Please, describe detailedly what difficulty you are in

    - Why doesnt this map server command generates this issue. Need to solve this issue. Let me know how to solve this error

rjshim, 2019-10-07:
@priyablue 

Hi, I just tested the same thing as what you wrote in the question 7 (terminal commands) and it worked fine for me. I briefly googled your problem and [this](https://answers.ros.org/question/303002/map_server-segfaults) might help you.

If it still doesn't work, I suggest you just reinstall OS and ROS 2 and test it again, or ask this question at [nav2_map_server](https://github.com/ros-planning/navigation2/tree/master/nav2_map_server)

Regards,
Ryan

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

closed 2019-10-15 by rjshim

priyablue, 2019-10-17:
Extremely sorry that there were no response from my end. The mail I
received got into spam folder.

I will look into the link shared by you and try. Will let you know.

Thanks for your response.