ROBOTIS-GIT/turtlebot3 #497: Robot goes crazy when running the pointop_key node

Pablowally, 2019-10-01:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ x Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ x] Gazebo

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ x] Ubuntu 16.04

5. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - I run the pointop_key node. It asks for coordinates, I input them. As soon as it finishes, it asks again, so I input new coordinates.
The robot starts spinning around and goes out of control.

Coordinates are: 
14,420
15,1,0

8. Copy and Paste your error message on terminal

    - No error message... The robot just spins around.

rjshim, 2019-10-07:
@Pablowally 

I tested it and I don't see any problems. Just in case, as shown in the video instruction, there shouldn't be any commas between input values (e.g., 14 4 20 instead of 14,4,20). If it still doesn't work, attach screenshots of your terminal and gazebo (or rviz) windows.

Regards,
Ryan

Pablowally, 2019-10-07:
Hi! I apologize, I realized 2 things:
The coordinates that give error are: 12 4 20 and then 15 1 0 (I do input them without commas)
It has nothing to do with me running the program twice: Problem is the code to find the location.

I made a little video that shows it: It only lasts 20 seconds. https://youtu.be/IVfzqDCTOqs
The problem comes with this part of the code (I added some informative prints):
```
if path_angle < -pi/4 or path_angle > pi/4:
                if goal_y < 0 and y_start < goal_y:
                    path_angle = -2*pi + path_angle
                elif goal_y >= 0 and y_start > goal_y:
                    path_angle = 2*pi + path_angle
            if last_rotation > pi-0.1 and rotation <= 0:
                rotation = 2*pi + rotation
                print("Rotation:"+str(rotation))
            elif last_rotation < -pi+0.1 and rotation > 0:
                rotation = -2*pi + rotation
                print("Rotation:"+str(rotation))
            move_cmd.angular.z = angular_speed * path_angle-rotation
            print("VELOCIDAD ENVIADA:"+str(move_cmd.angular.z))
```
I think that exactly at pi/4 the angle changes and the robot goes crazy, I'm still figuring out how to solve it tho.

Thanks a lot for your answer!

rjshim, 2019-10-07:
@Pablowally 

Then, would you test below and see if it works?

if path_angle < -pi/4 or path_angle >= pi/4:

Regards,
Ryan

Pablowally, 2019-10-07:
@rjshim Hi,

Nop, it gives the same problem. 
I put some prints inside the code, to understand it better:
I copy the prints and explain what I think is happening:

Every cycle I print the final sent w velocity, the taken rotation of the robot at the beginning, and then those prints if the conditions fullfill:
`if path_angle < -pi/4 or path_angle >= pi/4:
                if goal_y < 0 and y_start < goal_y:
                    path_angle = -2*pi + path_angle
                    print("New path angle:"+str(path_angle))
                elif goal_y >= 0 and y_start > goal_y:
                    path_angle = 2*pi + path_angle
                    print("New path angle:"+str(path_angle))
            if last_rotation > pi-0.1 and rotation <= 0:
                rotation = 2*pi + rotation
                print("New Rotation:"+str(rotation))
            elif last_rotation < -pi+0.1 and rotation > 0:
                rotation = -2*pi + rotation
                print("New Rotation:"+str(rotation))`

> control your Turtlebot3!
-----------------------
Insert xyz - coordinate.
x : position x (m)
y : position y (m)
z : orientation z (degree: -180 ~ 180)
If you want to close, insert 's'
-----------------------

| x | y | z |
15 1 0

**Note that the path angle is -0.77 rads (almost -pi/4)**, and the angular velocity sent to the robot is correct, -0.98 (it goes towards its objective)

> Path angle: Arctan 1.0-3.97601083436/15.0-11.9790386882
> Path angle (rads):-0.777902774971
> Rotation of the robot:0.257819212424
> Sent angular velocity:-1.0357219874
> --------New cycle--------
> Path angle: Arctan 1.0-3.97601083436/15.0-11.9790386882
> Path angle (rads):-0.777902774971
> Rotation of the robot:0.257819212424
> Sent angular velocity:-1.0357219874
> --------New cycle--------
> Path angle: Arctan 1.0-3.9762410355/15.0-11.9792694121
> Path angle (rads):-0.777979629675
> Rotation of the robot:0.258861016097
> Sent angular velocity:-1.03684064577
> --------New cycle--------
> Path angle: Arctan 1.0-3.97794449659/15.0-11.9823269021
> Path angle (rads):-0.778771986982
> Rotation of the robot:0.259961762178
> Sent angular velocity:-1.03873374916
> --------New cycle--------
> Path angle: Arctan 1.0-3.97927564565/15.0-11.987638636
> Path angle (rads):-0.779876236423
> Rotation of the robot:0.217420062098
> Sent angular velocity:-0.997296298521
> --------New cycle--------
> Path angle: Arctan 1.0-3.98076266744/15.0-11.995601549
> Path angle (rads):-0.781449120051
> Rotation of the robot:0.14640593253
> Sent angular velocity:-0.927855052581
> --------New cycle--------
> Path angle: Arctan 1.0-3.98182507938/15.0-12.0059802183
> Path angle (rads):-0.783357500416
> Rotation of the robot:0.0536818886229
> Sent angular velocity:-0.837039389039

**But suddenly here, the path angle reaches -pi/4 (=-0.78539816339), and then the **if** condition becomes true, the path angle changes, and the robot velocity goes to 5.55 (it starts going spinning the other way around)**

--------New cycle--------
Path angle: Arctan 1.0-3.98178485464/15.0-12.0203052936
Path angle (rads):-0.785748772357
Rotation of the robot:-0.0611981560114
New path angle:5.49743653482
Sent angular velocity:5.55863469083
--------New cycle--------
Path angle: Arctan 1.0-3.98117951241/15.0-12.0274436029
Path angle (rads):-0.786846516063
Rotation of the robot:-0.112237554188
New path angle:5.49633879112
Sent angular velocity:5.6085763453

After a while, it happens again but in the other side (robot is spinning with positive W, and it is reaching its path like that, but then the if condition is not fulfilled anymore, and the path angle goes back to the calculated one (without adding -2pi), and so it keeps rotating for ever.

> 
> --------New cycle--------
> Path angle: Arctan 1.0-4.01809914691/15.0-11.9934411148
> Path angle (rads):-0.787313665903
> Rotation of the robot:-1.4852933616
> New path angle:5.49587164128
> New Rotation:4.79789194558
> Sent angular velocity:0.697979695698
> --------New cycle--------
> Path angle: Arctan 1.0-4.00885381194/15.0-11.9945729881
> Path angle (rads):-0.785967940533
> Rotation of the robot:-1.40974686792
> New path angle:5.49721736665
> New Rotation:4.87343843926
> Sent angular velocity:0.623778927389
> --------New cycle--------
> Path angle: Arctan 1.0-3.99964044072/15.0-11.9963786752
> Path angle (rads):-0.784735043313
> Rotation of the robot:-1.3422945464
> New Rotation:4.94089076078
> Sent angular velocity:-5.72562580409

rjshim, 2019-10-07:
@Pablowally 

Thank you for the comment above. We are planning to update this example as [this](https://github.com/ROBOTIS-GIT/turtlebot3/blob/f69673e9951d4be6c7001c1b038860359861c040/turtlebot3_example/turtlebot3_example/turtlebot3_position_control/turtlebot3_position_control.py) in the Dashing version. Please refer to it if you need to use this example for your project or something now. We will fix this bug as soon as possible. 

Thank you very much,
Ryan

Pablowally, 2019-10-07:
@rjshim 

Thanks for your answer. 
That other solution works alright, but is a different concept! 

I will use it if I cannot fix this one, but this one looks nicer, it's a more "real" movement, don't you think?

BTW: I tryed commenting out the if condition, and then the problem happens when the robot has to move from 2 points with the same y coordinate, but different x.

Thanks!

rjshim, 2019-10-07:
@Pablowally 

Yeah, the main purpose of this package is to allow users to give relative target poses (e.g., move forward 10cm or turn 90 degrees clockwise from the current pose) so the path planning method is simpler than the original one. I don't know how you are going to use this example but if you just want to give global target pose commands, such as (10,1) on the global map, I recommend you make a map using the slam package and give goal poses using the navigation package. [ref](https://github.com/rjshim/turtlebot3_demo/blob/master/src/turtlebot3_demo.cpp)

Regards,
Ryan

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

Pablowally, 2019-11-04:
Hi,

I solved the problem. I attach the code for the first while loop here:

```
 while distance > 0.05:
            (position, rotation) = self.get_odom()
            x_start = position.x
            y_start = position.y
            path_angle = atan2(goal_y - y_start, goal_x - x_start)

            if(path_angle) < 0:
                path_angle = 2*pi+path_angle

            w = path_angle-rotation
            if w > pi:
                w = w - 2*pi
            move_cmd.angular.z = angular_speed * w

            distance = sqrt(pow((goal_x - x_start), 2) +
                            pow((goal_y - y_start), 2))
            move_cmd.linear.x = min(linear_speed * distance, 0.1)

            if move_cmd.angular.z > 0:
                move_cmd.angular.z = min(move_cmd.angular.z, 1.5)
            else:
                move_cmd.angular.z = max(move_cmd.angular.z, -1.5)

            last_rotation = rotation
            self.cmd_vel.publish(move_cmd)
            r.sleep()
        (position, rotation) = self.get_odom()
```

JaehyunShim, 2020-10-16:
I will close this thread as there have been no recent updates.

closed 2020-10-16 by JaehyunShim