ROBOTIS-GIT/turtlebot3 #498: Do we have way to get odom optimized by SLAM (cartographer)?

shakey02, 2019-10-02:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [*] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [*] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [*] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [*] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [*] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

9. Please, describe detailedly what difficulty you are in

Hello, I wonder if we can get the current odometry of the turtlebot optimized by SLAM system (cartographer ROS) since the system doesn't automatically send updated odom to our turtlebot.
Thanks

rjshim, 2019-10-07:
@shakey02 

You can see updated odom from the /tf topic (try rostopic echo /tf). Can I ask why you need to update specifically the /odom topic?

Regards,
Ryan

shakey02, 2019-10-07:
Since I move the robot on the carpet, when I use teleop to move the turtlebot for about 1 min in a small area, usually I cannot use pointop_key to return the robot to the origin (in real and on slam)

rjshim, 2019-10-08:
@shakey02 

Do you specific reasons why you need to use teleop to move your robot? I recommend you use the navigation pkg instead as [ref](https://github.com/rjshim/turtlebot3_demo/blob/master/src/turtlebot3_demo.cpp)

Regards,
Ryan

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

closed 2019-10-15 by rjshim