ROBOTIS-GIT/turtlebot3 #499: Bringup timeout

Lianstabity, 2019-10-03:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [X] Jetson Nano

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste your error message on terminal

... logging to /home/turtlebot2/.ros/log/04081e4e-e584-11e9-ae96-fcf8aee48a72/roslaunch-turtlebot2-10633.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.209:34701/

SUMMARY

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.218:11311

process[turtlebot3_core-1]: started with pid [10644]
process[turtlebot3_lds-2]: started with pid [10645]
process[turtlebot3_diagnostics-3]: started with pid [10646]
[INFO] [1570069469.005502]: ROS Serial Python Node
[INFO] [1570069469.056773]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1570069476.224376]: Requesting topics...
[ERROR] [1570069486.336193]: Write timeout: Write timeout
[ERROR] [1570069491.324457]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

9. Please, describe detailedly what difficulty you are in

Hi Robotis,

1. I update my OpenCR in recovery mode on Arduino IDE which shows jump_to_fw. 
2. I connected with my SBC, Jetson nano. I found SBC super slow when it boot up. Without OpenCR, my SBC become back to normal. 
3. It always can't found "/dev/ttyACM0" until I push the reset pin on OpenCR.
4. The Error Massage come from after Step 1 to 3.
I have 3 OpenCR, all of then have same issue. Also my SBC is new and clean setup.
Please have a see for this problem. Thanks

rjshim, 2019-10-07:
@Lianstabity 

We do not officially support the jetson version so I cannot give any help. @t-kitajima Have you also tried jetson-nano when you tested [this](https://www.youtube.com/watch?v=r8Src0LOORU)? Can you give him any advice?

t-kitajima, 2019-10-07:
@Lianstabity 

Hi,
We(Japan branch) have not tested with Jetson Nano.
However, this error may appear when the rosserial_python version is old.
```
[ERROR] [1570069491.324457]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
```
How about updating according to [this issue?](https://github.com/ROBOTIS-GIT/turtlebot3/issues/130)

rjshim, 2019-10-15:
As there have been no updates, I am closing this issue.

closed 2019-10-15 by rjshim