ROBOTIS-GIT/turtlebot3 #500: Strange camera_info

HowardWhile, 2019-10-05:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3

4. Which OS you installed in SBC?

    - [x] Raspbian

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)

Hi ROBOTIS

When I try to initial raspberry pi camera by command.

```shell
roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch
```

I found the camera parameter at `"turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml"` is a little weird. 

below is the parameters in the file.

```yaml
image_width: 640
image_height: 480
camera_name: camera
camera_matrix:
  rows: 3
  cols: 3
  data: [322.0704122808738, 0, 199.2680620421962, 0, 320.8673986158544, 155.2533082600705, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.1639958233797625, -0.271840030972792, 0.001055841660100477, -0.00166555973740089, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [329.2483825683594, 0, 198.4101510452074, 0, 0, 329.1044006347656, 155.5057121208347, 0, 0, 0, 1, 0]
```

the camera_matrix look like below

[image]

![](https://latex.codecogs.com/gif.download?\begin{bmatrix}f_x&0&c_x\\0&f_y&c_y\\0&0&1\end{bmatrix}=\begin{bmatrix}322.07&0&199.26\\0&320.86&155.25\\0&0&1\end{bmatrix})

Refer to the description in [opencv](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html), [image] is a principal point that is usually at the image center.

And Consider the image length and width is 640x480. 

(199.26, 155.25) is a little far from the image center (320, 240).

So I tried to recalculate the camera parameters follow [this tutorials](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration). The new parameters  as follows:

```yaml
image_width: 640
image_height: 480
camera_name: camera
camera_matrix:
  rows: 3
  cols: 3
  data: [509.002395, 0.000000, 327.562293, 0.000000, 506.358721, 245.259233, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.183372, -0.312891, -0.001562, 0.005458, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [521.060059, 0.000000, 331.136252, 0.000000, 0.000000, 521.592163, 244.694909, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

Now I can calculate the correct tag posture through this camera_info. 

The tag detection algorithm I use is [this](http://wiki.ros.org/apriltag_ros)

Best Regards, and provide these messages for you.

rjshim, 2019-10-07:
@HowardWhile 

Thank you very much. I tested as your advice and I've got a result similar to yours. We will discuss it in the team and update our package. Thank you so much for your contribution.

Regards,
Ryan

sunghowoo, 2024-11-01:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues, but as this issue has been open for a long time, we will close it for now. Currently, our team is supporting Noetic and Humble versions. Please try again with the latest version, and if you encounter the issue again, feel free to reopen it.

closed 2024-11-01 by sunghowoo