ROBOTIS-GIT/turtlebot3 #501: 360 Laser Distance Sensor LDS-01

Adel-kn, 2019-10-14:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?
- [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?
 [ ] ROS 1 Kinetic Kame
    

3. Which SBC(Single Board Computer) is working on TurtleBot3?
 [ ] Raspberry Pi 3
 

4. Which OS did you install in SBC?
 [ ] Raspbian

5. Which OS did you install in Remote PC?
 [ ] Ubuntu 16.04 LTS (Xenial Xerus)

6. Write down software version and firmware version
 ROS version 1.12.14

Hello everyone,
I added 3 layers more to my turtlebot3 and these layers are above the laser scanner (Laser scanner LDS-01).
The performance of the navigation and slam was affected because of the supporters which added to fix the layers.
I think  maybe one can  play with the resolution of the sensor to ignore  these supporters  
in such a way they do not exist, but I am not sure and have no idea how can one change this resolution.
If anyone has ideas to solve this problem please share it with me.
Best wishes

[image]

rjshim, 2019-10-15:
@Adel-kn 

TB3 uses the hls_lfcd_lds_driver package for sensor data from lidars as you can see [here](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_bringup/package.xml). Download the package from their github [page](https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver) and modify the minimum range to a value larger than the radius of the robot in the source code.

Test it and let us know the result,
Ryan

Adel-kn, 2019-10-15:
Hello @rjshim 
Thank you very much the problem was solved and no more detection of the supporters as obstacles.

Adel-kn, 2019-10-15:
@rjshim 
It still a small problem, that the robot still sees while navigation a few points around it as an obstacle although there is nothing.
Have you any idea why that happens?

rjshim, 2019-10-16:
@Adel-kn 

It is laser scan data noise so you need to use scan filters to remove it. 
I suggest you try [this](http://wiki.ros.org/laser_filters) and but we can really give any help with it.

Regards,
Ryan

closed 2019-10-16 by rjshim

Adel-kn, 2019-10-16:
Thanks Ryan