ROBOTIS-GIT/turtlebot3 #502: turtlebot3 Web Application (GUI)

Adel-kn, 2019-10-15:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello everyone, I should create a web application to control my robot from a web site, which represents a simple interface (GUI) to allow the normal user to deal with the robot without needing to type commands and enter inputs from the terminal. Basically, my application is a code (written in python) to perform a specific movement of the robot. This application asks the user to enter some parameters and depending on these parameters the robot act and moves. Does anyone have an idea of how one can perform this web app in a simple way? 
Thanks in advance

rjshim, 2019-10-16:
@Adel-kn 

We do not support such functions for the tb3 platform but there is a [reference](http://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-use-web-setting-tool) of your interest. 

Regards,
Ryan

Adel-kn, 2019-10-16:
Thanks Ryan

closed 2019-10-16 by rjshim