ROBOTIS-GIT/turtlebot3 #503: Difference in movement between the input value and the actual movement

Adel-kn, 2019-10-16:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello everyone,
I wrote two codes to derive my robot one is for linear movement and the other is for the rotational movement.
the problem is that the error is always increased by increasing the traveled distance (rotation /angle). That means if I enter 360 degrees rotation angle the robot will rotate almost 340 degrees (20 degrees error) and this error will higher when rotating for example 720 degrees and so on.
When I apply the code using "fake node" there is no errors.
Does anyone meet the same problem and have ideas to solve it?
Best wishes 
Below is my code (for the rotation movement):

#!/usr/bin/env python
import rospy
from geometry_msgs.msg import Twist
PI = 3.1415926535897
    
def move():
        print("Let's rotate your robot")
        speed = input("Input your speed (degrees/sec):")
        angle = input("Type your distance (degrees):")
        clockwise = input("Clockwise?: ") 
   
       
        angular_speed = speed*2*PI/360
        relative_angle = angle*2*PI/360
        if clockwise:
            vel_msg.angular.z = -abs(angular_speed)
        else:
            vel_msg.angular.z = abs(angular_speed)
        
        t0 = rospy.Time.now().to_sec()
        
        r=rospy.Rate(1000)
        current_angle = 0
        while(current_angle < relative_angle):
            velocity_publisher.publish(vel_msg)
            t1 = rospy.Time.now().to_sec()
            current_angle = angular_speed*(t1-t0)
            r.sleep()
    
    
        
        vel_msg.angular.z = 0
        velocity_publisher.publish(vel_msg)
      
    
if __name__ == '__main__':
     try:
         rospy.init_node('move', anonymous=True)
         velocity_publisher = rospy.Publisher('/cmd_vel', Twist, queue_size=10)
         vel_msg = Twist()
         move()
     except rospy.ROSInterruptException:
           pass

rjshim, 2019-10-17:
@Adel-kn 

Wheel slipping is the main reason ([ref](https://link.springer.com/article/10.1007/s10846-013-9871-1)) and localisation is needed in that case. Rather than trying to implement the algorithm by yourself, I suggest using SLAM (for making a map) and navigation (using the map) packages we provide ([ref](http://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#slam)).

Ryan  :)

Adel-kn, 2019-10-18:
helo Ryan,
Thanks for your answer. Actually I have already done many experiments using the SLAM and Navigation packages, but the task which I should implement is not related to them but it is only to enter some parameters to drive the robot in a specific way.

Adel-kn, 2019-10-18:
Hello Ryan,
I solved this problem by doing the next:
I did some experiments and found that the error was almost (-13 degree) therefore in the code I added this error to the angle entered by the user.
And because of this -13 degree was only for 360 degrees and it increases by increasing the angle I established a formula that increases this expected error 13 by increasing the angle:
 E.g. 13 for 360 
	26(13*2) for 720 (360*2) , 39 (13*3) for 1080 (360*3) and so on 
the formula that is being used:
 relative_angle = (angle+((angle/360)*13))*2*PI/360
where the term ((angle/360)*13)  represents the increased error by increasing the angle

rjshim, 2019-10-21:
@Adel-kn  
Good :) Let us know if you have any other questions. I will close this issue.  Ryan

closed 2019-10-21 by rjshim