ROBOTIS-GIT/turtlebot3 #504: Rosserial Error

jcotof, 2019-10-17:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [X ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [X ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ X] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

roslaunch turtlebot3_bringup turtlebot3_robot.launch 
   

8. Copy and Paste your error message on terminal

 [ERROR] [1571279560.041441]: Mismatched protocol version in packet: lost sync or rosserial_python is from different ros release than the rosserial client
Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)

9. Please, describe detailedly what difficulty you are in

Everything was working fine to this day in my burger, and nothing has changed, but now it is showing me the following error when I run the roslaunch turtlebot3_bringup turtlebot3_robot.launch command on my burger

Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)

rjshim, 2019-10-17:
@jcotof 

Would you redo SBC and OpenCR setups and see if it works? [ref](http://emanual.robotis.com/docs/en/platform/turtlebot3/setup/#setup)

Regards,
Ryan

jcotof, 2019-10-17:
1. We redo the SBC and OpenCR settings,but the problem persists.
2. We reinstalled the ROS and turtlebot packages and (on turtlebot) It works ok, but after I turned off and turned on the burger, the error persists.

rjshim, 2019-10-18:
@jcotof 

Please refer to [this](https://github.com/ROBOTIS-GIT/turtlebot3/issues/160#issuecomment-514896184) and let me know if it works.

Regards,
Ryan

jcotof, 2019-10-18:
It didn't work.

jcotof, 2019-10-20:
re-installed ROS on the Pi and everything is working fine now.

rjshim, 2019-10-21:
@jcotof Good :) I will close this issue. Ryan

closed 2019-10-21 by rjshim