ROBOTIS-GIT/turtlebot3 #505: Reg:Couldn't determine robot's pose associated with laser scan - burger

tatag4u, 2019-10-22:
Hi
I am getting the error while I am doing navigation topic in turtlebot3 burger using Ros kinetic with Ubuntu 16.04
[ERROR] [1571763280.542860289]: Couldn't determine robot's pose associated with laser scan
Help me to solve this error?

ROBOTIS-Will, 2019-11-07:
@tatag4u 
Hi,
Thank you for your inquiry.
While I'm awaiting from developers reply, could you please share us more detailed information?
For example, in which procedure you run the commands and what kind of error messages did you get from each terminal?
Thank you.

tatag4u, 2019-11-08:
Hi,
Thanks for your reply.
First I have created map using SLAM and it is stored in $HOME/map.yaml
**1)Remote PC: $ roscore**
... logging to /home/shiva/.ros/log/4962bb22-0242-11ea-a2c4-8cec4b6feb32/roslaunch-shiva-8415.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.16.28.67:36839/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [8425]
ROS_MASTER_URI=http://172.16.28.67:11311/

setting /run_id to 4962bb22-0242-11ea-a2c4-8cec4b6feb32
process[rosout-1]: started with pid [8438]
started core service [/rosout]

**2)Turtlebot3 (burger):$ roslaunch turtlebot3_bringup turtlebot3_robot.launch**
... logging to /home/pi/.ros/log/4962bb22-0242-11ea-a2c4-8cec4b6feb32/roslaunch-raspberrypi-1094.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.16.20.100:42211/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: odom
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://172.16.28.67:11311

process[turtlebot3_core-1]: started with pid [1120]
process[turtlebot3_lds-2]: started with pid [1121]
process[turtlebot3_diagnostics-3]: started with pid [1122]
[INFO] [1572526178.943060]: ROS Serial Python Node
[INFO] [1572526178.974476]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1572526181.121154]: Note: publish buffer size is 1024 bytes
[INFO] [1572526181.124335]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1572526181.140936]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1572526181.297942]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1572526181.308419]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1572526181.356831]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1572526181.367697]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1572526181.378821]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[... 182 lines cut ...]
[ INFO] [1573229873.705044711]: Using plugin "inflation_layer"
[ WARN] [1573229873.775901568]: Costmap2DROS transform timeout. Current time: 1573229873.7757, global_pose stamp: 1572526502.3082, tolerance: 0.5000
[ INFO] [1573229873.824769980]: Loading from pre-hydro parameter style
[ERROR] [1573229873.832147684]: Couldn't determine robot's pose associated with laser scan
[ INFO] [1573229873.877711506]: Using plugin "obstacle_layer"
[ INFO] [1573229873.880744097]:     Subscribed to Topics: scan
[ INFO] [1573229873.911904576]: Using plugin "inflation_layer"
[ INFO] [1573229873.987216365]: Created local_planner dwa_local_planner/DWAPlannerROS
[ INFO] [1573229873.994790930]: Sim period is set to 0.10
[ERROR] [1573229874.034628161]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229874.234476554]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229874.435480491]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229874.635490807]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229874.837085706]: Couldn't determine robot's pose associated with laser scan
[ WARN] [1573229874.875884254]: Costmap2DROS transform timeout. Current time: 1573229874.8758, global_pose stamp: 1572526502.3082, tolerance: 0.5000
[ERROR] [1573229875.037425516]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229875.237939488]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229875.437291454]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229875.638362809]: Couldn't determine robot's pose associated with laser scan
[ERROR] [1573229875.838407654]: Couldn't determine robot's pose associated with laser scan
[ WARN] [1573229875.875905753]: Costmap2DROS transform timeout. Current time: 1573229875.8758, global_pose stamp: 1572526502.3082, tolerance: 0.5000
      ,, ,,  ,,, .................
The error is repeated again.....

I run this on ROS on Ubuntu 16.04.

tatag4u, 2019-11-08:
Hi,
Thanks for your reply.
First I have created map using SLAM (map is stored in $HOME/map.yaml)
*1)Remote PC: $ roscore*
... logging to
/home/shiva/.ros/log/4962bb22-0242-11ea-a2c4-8cec4b6feb32/roslaunch-shiva-8415.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.16.28.67:36839/
ros_comm version 1.12.14
SUMMARY

PARAMETERS

   - /rosdistro: kinetic
   - /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [8425]
ROS_MASTER_URI=http://172.16.28.67:11311/

setting /run_id to 4962bb22-0242-11ea-a2c4-8cec4b6feb32
process[rosout-1]: started with pid [8438]
started core service [/rosout]

*2)Turtlebot3 (burger):$ roslaunch turtlebot3_bringup
turtlebot3_robot.launch*
... logging to
/home/pi/.ros/log/4962bb22-0242-11ea-a2c4-8cec4b6feb32/roslaunch-raspberrypi-1094.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.16.20.100:42211/
SUMMARY

PARAMETERS

   - /rosdistro: kinetic
   - /rosversion: 1.12.13
   - /turtlebot3_core/baud: 115200
   - /turtlebot3_core/port: /dev/ttyACM0
   - /turtlebot3_core/tf_prefix:
   - /turtlebot3_lds/frame_id: odom
   - /turtlebot3_lds/port: /dev/ttyUSB0

NODES
/
turtlebot3_core (rosserial_python/serial_node.py)
turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://172.16.28.67:11311

process[turtlebot3_core-1]: started with pid [1120]
process[turtlebot3_lds-2]: started with pid [1121]
process[turtlebot3_diagnostics-3]: started with pid [1122]
[INFO] [1572526178.943060]: ROS Serial Python Node
[INFO] [1572526178.974476]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1572526181.121154]: Note: publish buffer size is 1024 bytes
[INFO] [1572526181.124335]: Setup publisher on sensor_state
[turtlebot3_msgs/SensorState]
[INFO] [1572526181.140936]: Setup publisher on firmware_version
[turtlebot3_msgs/VersionInfo]
[INFO] [1572526181.297942]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1572526181.308419]: Setup publisher on cmd_vel_rc100
[... 214 lines cut ...]
*[ERROR] [1573229874.034628161]: Couldn't determine robot's pose associated
with laser scan [ERROR] [1573229874.234476554]: Couldn't determine robot's
pose associated with laser scan [ERROR] [1573229874.435480491]: Couldn't
determine robot's pose associated with laser scan [ERROR]
[1573229874.635490807]: Couldn't determine robot's pose associated with
laser scan [ERROR] [1573229874.837085706]: Couldn't determine robot's pose
associated with laser scan [ WARN] [1573229874.875884254]: Costmap2DROS
transform timeout. Current time: 1573229874.8758, global_pose stamp:
1572526502.3082, tolerance: 0.5000 [ERROR] [1573229875.037425516]: Couldn't
determine robot's pose associated with laser scan [ERROR]
[1573229875.237939488]: Couldn't determine robot's pose associated with
laser scan [ERROR] [1573229875.437291454]: Couldn't determine robot's pose
associated with laser scan [ERROR] [1573229875.638362809]: Couldn't
determine robot's pose associated with laser scan [ERROR]
[1573229875.838407654]: Couldn't determine robot's pose associated with
laser scan [ WARN] [1573229875.875905753]: Costmap2DROS transform timeout.
Current time: 1573229875.8758, global_pose stamp: 1572526502.3082,
tolerance: 0.5000*
,, ,, ,,, .................
The error is repeated again.....

I run this on ROS on Ubuntu 16.04.

Thank you
Siva

tatag4u, 2019-11-08:
Hi

I have attached rqt_graph.

Thank you
Siva

ROBOTIS-Will, 2019-11-11:
@tatag4u 
Hi,
Thank you for attaching the log.
Have you performed Initial Pose estimation written on eManual?
[http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#estimate-initial-pose](http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#estimate-initial-pose)
After map and TurtleBot3 is loaded on the RViz screen, there's a **`2D Pose Estimate`** button on top so that you can locate the initial pose of the actual robot in the map.
This has to be done before running the navigation.
Thank you.

tatag4u, 2019-11-11:
Hi

I have done 2D Pose estimate, but not exactly match with map(very small
offset is present between given map to lidar data).

With Regards
G.Sivasankar

tatag4u, 2019-11-12:
Hi

Moreover,that error occurred while loading rviz from navigation.launch file
itself(not during navigation to goal estimate).The same step while I am
doing it in simulation, turtlebot3 is moving to goal.

I have done investigation by calling the launch files(in navigation
.launch) individually. That error is created from amcl.launch file.

Thanks
G.Sivasankar

tatag4u, 2019-12-19:
Dear Sir

I think transformation is not correct for odom.

What I have shown in attachment is correct transformation ?

Please help me to correct it.

ROBOTIS-Will, 2020-01-07:
@tatag4u 
I'm sorry for the late reply. Could you point me where I can get the transformation and rqt_graph attachments? Thank you.

ROBOTIS-Will, 2020-05-11:
This thread will be closed as there wasn't recent activity. Please feel free to reopen when necessary.

closed 2020-05-11 by ROBOTIS-Will

onurcankaraca, 2021-08-06:
> Hi
> I am getting the error while I am doing navigation topic in turtlebot3 burger using ROS Kinetic on Ubuntu 16.04
> [ERROR] [1571763280.542860289]: Couldn't determine robot's pose associated with laser scan
> Help me to solve this error? (More specifically while running amcl.launch)
> But while doing slam.launch , it is working.

did you solve this problem? if u solved it can you share how you did it ?

Sivasankar-Ganesan, 2021-08-06:
I solve the problem by installing the dependency package and increase time
tolerance.

onurcankaraca, 2021-08-06:
> I solve the problem by installing the dependency package and increase time tolerance.
> […](#)
> -- [image: animated-india-flag-image-0004] With Regards G.Sivasankar Assistant Professor(Sr) Department of ECE Mepco Schlenk Engineering College, Sivakasi-626005 Ph.+91-4562-235429,09791562538

Hi;
Firstly thanks for the reply. I'll try it now but which dependency do you mean?