ROBOTIS-GIT/turtlebot3 #506: Problem with Raspberry pi camera

singingBuck, 2019-10-24:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

  I followed the manual step by step - [HERE](http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#raspberry-pi-camera)

8. Copy and Paste your error message on terminal

  Invoking "make cmake_check_build_system" failed

9. Please, describe detailedly what difficulty you are in

    Hello, everyone, I am dealing with this kind of problem while I am trying to make work raspberry 
    pi camera on Turtlebot3 - burger. The error message show up here is output after using 
   catkin_make command. It is followed by error message telling me that raspicam_node cannot be 
   found in src folder of my catkin_ws. I followed every step as is written in manual, but although 
   everything seems legit, the folders exist a the launch files, it's not working.

Thank for you response, 
Have a nice day.

ROBOTIS-Will, 2019-11-07:
@singingBuck 
Hi,
Please make sure to download the raspicam_node package in catkin_ws\src folder by using below command before the `git clone ....` command.
`cd ~/catkin_ws/src`
Thank you.

closed 2020-01-22 by ROBOTIS-Will

ROBOTIS-Will, 2020-01-22:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.