ROBOTIS-GIT/turtlebot3 #507: SLAM works badly with Carrier Wheels

shakey02, 2019-10-25:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

9. Please, describe detailedly what difficulty you are in

I recently built the carrier using tuetlebot3 waffle pi. Everything works fine until I changed for big wheels for Carrier. When I use SLAM to build map, it seems every time when the Cartographer ROS SLAM do loop closure (if it does so when the trajectory color changed from shallow blue to dark blue) trajectory shrank and before the shrinking the map seems correct. I wonder if it because these are some config about the gain factor from wheel radius?

Thank you very much

ROBOTIS-Kayman, 2019-10-28:
Hi,

If you changed the wheels, you have to modify the OPENCR firmware and upload to Turtlebot3 by using arduino IDE. 
Please refer to [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#alternative-arduino-ide).

- load the firmware for turtlebot3_waffle.
- modify [the variables](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#alternative-arduino-ide) for you robot.
- uplaod the firmware to turtlebot3.

Regards,
Kayman

shakey02, 2019-10-29:
Thanks a lot. BTW,do you have some suggestion on carrier config or moving control? Since it's a little bit too high for turtlebot to be stable while moving for carrier?

closed 2019-11-01 by shakey02