ROBOTIS-GIT/turtlebot3 #508: Lookup would require extrapolation into the future, for base_scan and odom in simulation mode

ribinmathew, 2019-10-29:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
   
2. Which ROS is working with TurtleBot3?

    - [ ] ROS 2 Dashing Diademata

3. Which SBC(Single Board Computer) is working on TurtleBot3?
     i am running it in my pc 

4. Which OS you installed in SBC?
    dont have one   
5. Which OS you installed in Remote PC?
  18.04
6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

i am running the turtlebot3 simulation in my pc,  the tf between base_scan and odom frame is keeps on missing...set every node to use_sim_time true, 

8. Copy and Paste your error message on terminal

[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 567.80200 but the latest data is at time 567.80000, when looking up transform from frame [base_scan] to frame [odom]

9. Please, describe detailedly what difficulty you are in

    -the tf is breaking in between kind of jumping, the laser scan keeps on blinking (on/off) (status ok  and error tf not found) in rviz

ROBOTIS-Will, 2019-11-07:
@ribinmathew 
This issue is caused by not having a message_filter in RViz2.
This will be resolved in ROS2 Eloquent.
Thank you.

mstuettgen, 2019-11-20:
Hi there,

I just tried it out with eloquent. Still getting this issue. My scenario:

sudo apt install ros-eloquent-desktop
sudo apt install ros-eloquent-gazebo-ros-pks

git clone -b dashing-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
git clone -b dashing-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git

colcon build

Then, running everything with use_sim_time true in all nodes, I still get RVIZ2 errors, e.g.:

[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 773,42800 but the latest data is at time 773,39800, when looking up transform from frame [base_scan] to frame [odom]

I have checked all parameters of all running nodes:
/camera_driver:
  use_sim_time
/gazebo:
  publish_rate
  use_sim_time
/launch_ros:
  use_sim_time
/robot_state_publisher:
  robot_description
  use_sim_time
/rviz2:
  use_sim_time
/turtlebot3_diff_drive:
  use_sim_time
/turtlebot3_imu:
  use_sim_time
/turtlebot3_joint_state:
  use_sim_time
/turtlebot3_laserscan:
  use_sim_time

in all of them "use_sim_time" was set to true. Any ideas?

ROBOTIS-Will, 2020-05-11:
This thread will be closed as there wasn't recent activity. Please feel free to reopen when necessary.

closed 2020-05-11 by ROBOTIS-Will