ROBOTIS-GIT/turtlebot3 #509: OpenCR corrupts Ubuntu

renno-bih, 2019-10-30:
Hello all,

I was having very strange issues lately with OpenCR and multiple computers. My Ubuntu will randomly become corrupted and will not boot. After reinstalling Ubuntu multiple times I found source of this problem but I don't know how to fix it.

Steps to reproduce problem:
 - Disconnect battery power from openCR (Dynamixel motor are not powered)
 - Turn on miniPC
 - Ubuntu gets corrupted on boot (not always but most of times)

This error is shown and computer freezes
```
/dev/mmcblk1p2: recovering journal
/dev/mmcblk1p2: Clearing orphaned inode 1322635 (uid=1000, gid=1000, mode=0100664, size=19725)

```

Looks like OpenCR is sending something to USB and it corrupts Ubuntu on boot. **This only happens if motors are disconnected or not powered.** If everything is powered everything is working properly.

I'm using the latest `turtlebot3_with_open_manipulator_core `firmware.

Any idea how to fix this. It is very frustrating to to spend whole day on installing ROS and then Ubuntu gets corrupted.
I had this problem with two copmuters: AK3V MiniPC and IntelUp board.

renno-bih, 2019-11-19:
Any updates regarding this issue?

ROBOTIS-Will, 2020-12-09:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-09 by ROBOTIS-Will