ROBOTIS-GIT/turtlebot3 #510: Turtlebot don't move when I publish in cmd_vel

bestkid1234, 2019-10-30:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 --I am not sure what should I write here

7. Write down the commands you used in order

    1. `ssh pi@192.168.0.201`
    2. `roslaunch turtlebot3_bringup turtlebot3_robot.launch `
    3. (on remote pc): `roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch`

8. Copy and Paste your error message on terminal

    - none

9. Please, describe detailedly what difficulty you are in

    -I am trying to move turtlebot from keyboard but it don't respond to commands. The remote PC is properly connected to robot. In file `.bashrc` variables are :  `ROS_HOSTNAME=192.168.0.106 `, `ROS_MASTER_URI=http://192.168.0.201:11311`
   -When I tried to publish with `rostopic pub /cmd_vel [...] `it is publishing but the robot do not move.
   -From other PC it works and I can not figure out why I can not move robot from my PC

ROBOTIS-Will, 2019-11-07:
@bestkid1234 
Please check if you're getting proper cmd_vel topics from the RPi side.
Use ssh to connect RPi on TurtleBot3 and `rostopic echo /cmd_vel` and see if correct values are received.
If cmd_vel is not normal, it could be the network issue.
If received cmd_vel is normal from RPi, please update or recover the OpenCR firmware.
Thank you.

bestkid1234, 2019-11-07:
> @bestkid1234
> Please check if you're getting proper cmd_vel topics from the RPi side.
> Use ssh to connect RPi on TurtleBot3 and `rostopic echo /cmd_vel` and see if correct values are received.
> If [...]

Hello! Thank you for reply! So, I tried to publish from my computer in `cms_vel` topic but when I `echo` it I can't see anything publishing. But if someone else is publishing, I can see the correct values if I `echo` the `cmd_vel` topic.

ROBOTIS-Will, 2019-11-25:
@bestkid1234 Are you using multiple PC as a remote?
Is ROS master is running on 192.168.0.201?
What do you see if you enter "rostopic list" command on your terminal when TB3 is online?

bestkid1234, 2019-11-26:
`~$ rostopic list 
/battery_state
/cmd_vel
/cmd_vel_rc100
/diagnostics
/firmware_version
/imu
/joint_states
/magnetic_field
/motor_power
/odom
/raspicam_node/image/compressed
/raw_image
/raw_image/compressed
/raw_image/compressed/parameter_descriptions
/raw_image/compressed/parameter_updates
/raw_image/compressedDepth
/raw_image/compressedDepth/parameter_descriptions
/raw_image/compressedDepth/parameter_updates
/raw_image/theora
/raw_image/theora/parameter_descriptions
/raw_image/theora/parameter_updates
/republish/compressed/parameter_descriptions
/republish/compressed/parameter_updates
/reset
/rosout
/rosout_agg
/rpms
/scan
/sensor_state
/sound
/tf
/version_info
`

bestkid1234, 2019-11-26:
> @bestkid1234 Are you using multiple PC as a remote?
> Is ROS master is running on 192.168.0.201?
> What do you see if you enter "rostopic list" command on your terminal when TB3 is online?

```
~$ rostopic list 
/battery_state
/cmd_vel
/cmd_vel_rc100
/diagnostics
/firmware_version
/imu
/joint_states
/magnetic_field
/motor_power
/odom
/raspicam_node/image/compressed
/raw_image
/raw_image/compressed
/raw_image/compressed/parameter_descriptions
/raw_image/compressed/parameter_updates
/raw_image/compressedDepth
/raw_image/compressedDepth/parameter_descriptions
/raw_image/compressedDepth/parameter_updates
/raw_image/theora
/raw_image/theora/parameter_descriptions
/raw_image/theora/parameter_updates
/republish/compressed/parameter_descriptions
/republish/compressed/parameter_updates
/reset
/rosout
/rosout_agg
/rpms
/scan
/sensor_state
/sound
/tf
/version_info
```

And yes, I am using more PC for remote. But not more that one at once.

bestkid1234, 2019-11-26:
> @bestkid1234 Are you using multiple PC as a remote?
> Is ROS master is running on 192.168.0.201?
> What do you see if you enter "rostopic list" command on your terminal when TB3 is online?

`192.168.0.201` is the robot's IP.

bestkid1234, 2019-11-29:
```
export ROS_MASTER_URI=http://192.168.0.200:11311
export ROS_HOSTNAME=192.168.0.104
export TURTLEBOT3_MODEL=burger

source /opt/ros/kinetic/setup.bash
source /home/ovidiu/catkin_ws/devel/setup.bash
```

Here is how my `.bashrc` file looks.

ROBOTIS-Will, 2019-12-04:
@bestkid1234 
It looks like you are using different ROS Master for TurtleBot3 and PC.
For both of them, ROS_MASTER_URI should be http://192.168.0.200:11311 which is the IP address of your PC.
Please refer to the network configuration section in eManual.
http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration

closed 2020-01-22 by ROBOTIS-Will

ROBOTIS-Will, 2020-01-22:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

stevenpandiri, 2020-04-20:
Hello Robotis-Will,

I have a very similar issue that when published on cmd_vel tutlebot3(burger) does not move. 

My configuration:

1. Created 1 docker container with ubuntu and ROS kinetic
2. Running ROS master and turtlebot3_core inside the same (single) container. 
3. turtlebot3_core connects to the master
4. rostopic list - lists all the topics correctly
5. when i publish on cmd_vel, turtlebot does not move (rostopic pub -1 /cmd_vel geometry_msgs/Twist -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]')
6. Response on publish - publishing and latching message for 3.0 seconds

My questions: 
Is it correct to run ROS master, turtlebot3_core on the same machine? and 
Also use rostopic to publish to turtlebot3_core from the same container?

I do understand that the setup mentioned in the manual is different but our requirement is such that we cannot use separate SBCs(computers) for remotePC and turtlebot PC

Thanks in advance for your responses

PranaliDesai, 2021-01-18:
I had the same issue and found a video - https://youtu.be/2WBDRbYmE7A

This video made sense.

Hope this helps