ROBOTIS-GIT/turtlebot3 #512: Determine transformation to camera

florianspy, 2019-11-06:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

   Looking at the transformations from the file turtlebot3_waffle_pi.urdf.xacro regarding the camera_link and camera_joint located below of turtlebot3_description/urdf, i would like to know how those transformations got calculated and if mount my camera on different place if there is a easy way to measure / determine them.

ROBOTIS-Will, 2019-11-07:
@florianspy 
Hi,
It is recommended to refer to the URDF tutorials below.
The `xyz` are actual distance in meter whereas `rpy` are in radian.
[http://wiki.ros.org/urdf/Tutorials](http://wiki.ros.org/urdf/Tutorials)
Thank you.

florianspy, 2019-11-10:
Dear Robotis-Will,
I guess i asked my question the wrong way i would like to know how those distances got determinated. 
So have they been measure with a specific device or is there a calibration procedure which outputs the distances needed for the transformation?

ROBOTIS-Will, 2019-11-11:
@florianspy ,
I'm sorry for the misunderstanding.
The parameters we provide in the code are precisely acquired from CAD tools where we can assemble parts and get the transformation data.
Thank you.

florianspy, 2019-11-11:
Dear Robotis-Will,
thank you for your answer, may i ask if there is information available regarding a transformations of the D435 Camera. We want to use it instead of the Raspberry PI Camera.

ROBOTIS-Will, 2019-11-12:
@florianspy 
Unfortunately, we do not have the transformation data for Realsense D435 on TurtleBot3 as it may vary by assembly method.
However, you can download the CAD(.step) from Intel site.
[https://www.intel.com/content/www/us/en/support/articles/000026841/emerging-technologies/intel-realsense-technology.html](https://www.intel.com/content/www/us/en/support/articles/000026841/emerging-technologies/intel-realsense-technology.html)
Thank you.

florianspy, 2019-11-12:
Dear Robotis-Will, 
So may i ask if you recommend any method / device for determine the transformation to the mounting place on the robot, e.g. using a ruler?

ROBOTIS-Will, 2019-11-25:
@florianspy I'm sorry for the delayed response.
If you don't have any 3D tools to simulate the assembly, you can use / calculate the transformation of the camera from the parent link. But for more accurate data, it is recommended to use 3D tools so that you can virtually assemble the custom designed frame that will hold the camera and measure the distance and angle from the parent link.

florianspy, 2019-12-03:
Dear Robotis-Will thank you for your reply. Unfortunately we have a custom design for mounting the camera, therefore we need to measure the respective transforms on the robot itself and cannot use a 3D model

ROBOTIS-Will, 2019-12-12:
@florianspy As long as your measurement is precise enough(I'd say in millimeter[mm]), you should be fine. Let me know if you need more information regarding the measurement reference points.

closed 2020-01-03 by ROBOTIS-Will

akshay-30, 2021-08-02:
May I know how we can obtain transforms of the bot? I want to obtain frame ids, similar to using hector_trajectory_server used in ROS