ROBOTIS-GIT/turtlebot3 #513: Turtlebot3 Waffle_Pi not moving in teleop_key demo or any other mode

JTMurley, 2019-11-06:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Melodic Morenia

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel NUC

4. Which OS you installed in SBC?

    - [ ] Ubuntu Server 18.04 or later

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 18.04 LTS (Bionic Beaver) through VM

6. Write down software version and firmware version

    - Software version: [1.12.13]
    - Firmware version: [1.2.2]

7. Write down the commands you used in order

Remote PC

roscore

Turtlebot3

roslaunch turtlebot3_bringup turtlebot3_core_launch

Remote PC

export TURTLEBOT3_MODEL=waffle_pi
roslaunch turtlebot3_bringup turtlebot3_remote.launch
roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

8. Copy and Paste your error message on terminal

    No Error

9. Please, describe detailedly what difficulty you are in

    Getting the Turtlebot3 to move with anything besides the basic OPENCR board buttons and the master slave remote control.

I'm pressing all of the buttons, W,A,S,D with the robot not moving at all. The OPENCR board is on and power by the battery 

POWER LED: ON (green)
Arduino LED: blinking (green)
USER4 LED: ON (green)
USER1 LED: blinking rapidly (green)
USER2 LED blinking every 5s (green)

ROBOTIS-Will, 2019-11-07:
@JTMurley 
Hi,
First of all, please make sure that you are pressing A,S,D,W,X keys while the turtlebot3_teleop_key running terminal is focused.
Also, please run the `rostopic echo /cmd_vel` from the NUC on TurtleBot3 to see if it is receiving proper topic from Remote PC.
Thank you.

JTMurley, 2019-11-09:
@ROBOTIS-Will 
Hi,
I managed to fix it, i'm unsure what the actual problem was, but its fixed now.
Cheers,
Jack

ROBOTIS-Will, 2019-11-12:
@JTMurley 
Thanks for the reply! Have fun with TurtleBot3 :)

closed 2019-11-12 by ROBOTIS-Will