ROBOTIS-GIT/turtlebot3 #514: Turtlebot does not power on anymore

bestkid1234, 2019-11-07:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    [none]

8. Copy and Paste your error message on terminal

    [none]

9. Please, describe detailedly what difficulty you are in

    I played with my turtlebot today and it got hit into a cardboard box. The console has blocked so I turn off the robot manually from the main board. After that when I tried to turn on it doesn't work. Just lighting up "Power" LED and the "Arduino" LED is flashing. But it does not connect to wifi or starting to rotate the sensor.

ROBOTIS-Will, 2019-11-08:
@bestkid1234 
Hi,
You might have damaged SBC or OpenCR.
I'd recommend to check for any visual damages on components first.
Then, please try OpenCR firmware recovery on Arduino IDE and see if it resolves the issue.
Thank you.

bestkid1234, 2019-11-26:
I sent my board card to warranty... so thank you anyway :+1:

closed 2019-12-04 by ROBOTIS-Will