ROBOTIS-GIT/turtlebot3 #515: Update launch files for ROS2 eloquent

AndreasAZiegler, 2019-11-08:
1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [x] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - See [here](https://github.com/ros-planning/navigation2/issues/1312#issuecomment-551199392)

8. Copy and Paste your error message on terminal

    - See [here](https://github.com/ros-planning/navigation2/issues/1312#issuecomment-551199392)

9. Please, describe detailedly what difficulty you are in

    - According to a comment [here](https://github.com/ros-planning/navigation2/issues/1312#issuecomment-551199392), the launch files seem not to be adjusted for ROS2 eloquent.

ROBOTIS-Will, 2019-11-12:
@AndreasAZiegler 
Thank you for reporting the issue. We'll review the issue and get back to you.
Thanks!

testkit, 2019-11-27:
I'm using Waffle, and also expect the support for ROS2 Eloquent.

rjshim, 2019-11-27:
@testkit 

Because some dependencies (cartographer_ros, nav2_bringup, robot_state_publisher, and joint_state_publisher) packages do not support Eloquent yet, we might start to work on it mid December or early January next year.

Ryan

testkit, 2019-11-27:
@rjshim, thanks for your information!

ninamwa, 2020-01-30:
@rjshim Is there any updates on this issue? :-)

heickrd, 2020-02-21:
@rjshim I am also getting this error. Is there any update on the launch files?

JaehyunShim, 2020-10-16:
I will close this thread as there have been no updates. Let me know if it still doesn't work.

closed 2020-10-16 by JaehyunShim