ROBOTIS-GIT/turtlebot3 #516: install the opencr package via board manager

techjjun, 2019-11-08:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [o ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x ] etc (not using)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ o] etc (nvidia xavier board , jetson tx2)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [o ] etc (Ubuntu 18.04 LTS)

5. Which OS you installed in Remote PC?

    - [o ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x] latest
    - Firmware version: [x.x.x] latest

7. Write down the commands you used in order

    - install the opencr package via board manager

8. Copy and Paste your error message on terminal

    - Tool opencr_tools is not available for your operating system.

9. Please, describe detailedly what difficulty you are in

    - I installed Arduino IDE and want to use Opencr package with nvidia Xavier board.
[image]

what operating system is wrong with this?

ROBOTIS-Will, 2019-11-11:
@techjjun 
Hi,
Unfortunately, current OpenCR board manager package does not support arm based SBCs.
Thank you.

techjjun, 2019-11-11:
Then also the raspberry pi3 can't install package??

Thanks

ROBOTIS-Will, 2019-11-12:
@techjjun 
Yes, this is why we provide the binary image for OpenCR firmware.
[http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#recommended-shell-script](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#recommended-shell-script)

closed 2019-12-12 by ROBOTIS-Will

ycean, 2021-08-25:
Hi, I would like to update the OpenCR firmware, for my customs robot platform, is there any clue for it? Because i am having the same problem.

ROBOTIS-Will, 2021-08-30:
Hi @ycean 
You may use your PC or laptop to upload the firmware directly from Arduino IDE.
Or if you have created a binary file (such as .bin) from Arduino IDE, you may use the downloader to upload the firmware
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#downloader
Thanks.

ycean, 2021-08-30:
Hi,

Thanks for the quick reply. May I consult with you regarding the firmware, or require steps that I need to do in order to use OpenCR board for my custom robot platform.

As I am actually unsure how is the schematic of robot control communication among the remote pc-sbc-openCR.

regards,

--

경북대학교  기계공학부 자동제어연구실
박사과정 양정연

=======================================================================

Kyungpook National University
System Control & Automation lab.

Ph.D. course.
Yeoh Chin Ean

CellPhone : +82-10-6397-0797
E-mail : ***@***.***
=======================================================================

https://emanual.robotis.com/docs/en/parts/controller/opencr10/#downloader

ROBOTIS-Will, 2021-08-30:
@ycean 
I'm sorry about deleting your reply as it seems accidently included with some personal information.
Here's your reply I copied.
As your inquiry doesn't seem to be an issue of TurtleBot3, I'd recommend creating a post in ROBOTIS community so that you can get help from other developers as well.
https://community.robotis.us/

> Hi,
> 
> Thanks for the quick reply. May I consult with you regarding the firmware, or require steps that I need to do in order to use OpenCR board for my custom robot platform.
> 
> As I am actually unsure how is the schematic of robot control communication among the remote pc-sbc-openCR.
> 
> regards,

ycean, 2021-08-30:
I didn't realize that my reply would appear on the forum too. Thanks with noted.

regards

--

경북대학교  기계공학부 자동제어연구실
박사과정 양정연

=======================================================================

Kyungpook National University
System Control & Automation lab.

Ph.D. course.
Yeoh Chin Ean

CellPhone : +82-10-6397-0797
E-mail : ***@***.***
=======================================================================

https://community.robotis.us/

ycean, 2021-08-30:
Thank with noted.

regards,
Yeoh

ROBOTIS-Will, 2021-08-30:
@ycean 
Not a problem.
Since people in this thread receives reply messages, please create a new thread in the aforementioned forum and start from fresh.