ROBOTIS-GIT/turtlebot3 #517: Multi-turtlebot navigation in real life environment

randyd55, 2019-11-22:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - [turtlebot] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - [Remote PC] roslaunch turtlebot3_navigation turtlebot3_navigation.launch 
       map_file:=$HOME/conference.yaml

8. Copy and Paste your error message on terminal
    - N/A

9. Please, describe detailedly what difficulty you are in

    - I'm trying to run through the navigation tutorial found here http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#navigation
but with multiple turtlebots. I know I need to configure the namespaces somehow so that the appropriate topics are published/subscribed to but I'm pretty lost as to how to do that. I thought maybe setting the ROS_NAMESPACE environment variable in each would be enough, but rviz still doesn't seem to be subscribing to the appropriate topics when I do that. Any direction/help would be appreciated.

ROBOTIS-Will, 2019-11-25:
@randyd55 
Please refer to the following eManual regarding running multiple TurtleBot3 within a single map.
[http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#virtual-slam-by-multiple-turtlebot3s](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#virtual-slam-by-multiple-turtlebot3s)
You'll be able to find how to set a different namespace on each robot.
Thank you.

randyd55, 2019-11-25:
I tried using this to help but I am still having trouble translating it to work for a real life simulation. For instance when I run the following commands:

[turtlebot] ROS_NAMESPACE=tb1 roslaunch turtlebot3_bringup turtlebot3_robot.launch

[remote PC] ROS_NAMESPACE=tb1 roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/conference.yaml set_base_frame:=tb1/base_footprint set_odom_frame:=tb1/odom set_map_frame:=tb1/map

rviz doesn't seem to be getting the map data. When I run rqt graph, it shows the rviz node subscribed to /map while the map_server node is publishing to tb1/map, so I tried to manually set the topic in rviz to subscribe to tb1/map instead, and rqt_graph now shows rviz subscribing to tb1/tb1/map and still no map appears.

ROBOTIS-Will, 2020-01-31:
#536 provides TF view of the issue when loading 2 TurtleBot3 with using different ROS_NAMESPACE.

randyd55, 2020-01-31:
@ROBOTIS-Will When I try to use the multi_robot name parameter I get this warning in terminal:

[ WARN] [1580503568.458021779]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100728 timeout was 0.1.

And many errors in rviz similar to this:

No transform from [base_footprint] to [map]

And the tf tree looks like this:

[image]

randyd55, 2020-01-31:
@ROBOTIS-Will I managed to get the tree looking like this by passing in a multi_robot_name to the robot state publisher launched by turtlebot3_navigation.launch:

[image]

But the map frame is still completely missing. I think some argument needs to be passed into the acml node that is launched, but I can't seem to figure it out. Any ideas?

ROBOTIS-Will, 2020-03-06:
@randyd55 
I'm sorry for the late reply.

For the amcl scan topic, you need to use /tb3_0/scan so it can get the LaserScan from tb3_0 properly.
When running the turtlebot3_navigation.launch, try appending scan_topic:=/tb3_0/scan and see if it corrects your issue.
You should also set the LaserScan Topic in RViz to /tb3_0/scan.

ROBOTIS-Will, 2020-04-21:
This issue will be closed as there was no activity on this thread. Please feel free to reopen the issue whenever needed.

closed 2020-04-21 by ROBOTIS-Will

shubham-shahh, 2021-08-17:
Hi @ROBOTIS-Will  I am facing a similar problem, posted [here](https://answers.ros.org/question/384446/multi-robot-turtlebot3-navigation-issues-with-laser-data-and-map-frame/) please look into this

thanks

Zuhair-A-Ahmed, 2024-04-14:
> Hi @ROBOTIS-Will I am facing a similar problem, posted [here](https://answers.ros.org/question/384446/multi-robot-turtlebot3-navigation-issues-with-laser-data-and-map-frame/) please look into this
> 
> thanks

Me too. Please, can you inform us with a solution.